skaarj1989/mWebSockets #3: Websocket disconnect

hendrikjvr, 2018-11-06:
I am using the websocket client on SAMD21 with a python script as websocket server on pc. What happens if the websocket is stopped or disconnected from a server side? How does the code handle this? Does the code handle this with an error? onClose is not triggered when I disconnect on server side - is this normal?

I am running on SAMD21 with W5500.

skaarj1989, 2018-11-06:
I did do some tests:
Server: node.js on PC (simple.js in examples directory)
Client: arduino mega2560 (simple-client.ino example)

Once I stopped the server, client was disconnected

```
Initializing ... 
[Line #0] HTTP/1.1 101 Switching Protocols
[Line #1] Upgrade: websocket
[Line #2] Connection: Upgrade
[Line #3] Sec-WebSocket-Accept: +byAJz6l9TL5qoP4wK4Jw8odnY8=
[Line #4] 
Connected
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=26, MASK=1b334c1b
Hello from Arduino client!
TX BYTES = 32
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=18, MASK=None
Hello from Node.js
Received: Hello from Node.js
Disconnected
```
I suggest you to check debug output on your serial monitor.
Also check browser client reaction when server disconnects (maybe your python client is broken?)

hendrikjvr, 2018-11-12:
Thank you, works with node.js!

closed 2018-11-12 by hendrikjvr