skaarj1989/mWebSockets #4: ESP8266 webSocket client client connection when desired

denden4444, 2018-12-24:
Hello 

This looks like a great library :-) 
Is it possible to enable the : `client.listen();`
only when required for example when wanting to establish a short connection to transfer a small amount of data and once done close it ?

If yes then can the following : 
` client.setOnOpenCallback(onOpen);
  client.setOnCloseCallback(onClose);
  client.setOnMessageCallback(onMessage);
  
  client.open("host", 3000);`

Also be invoked at the same time instead of at start(){} ?

Thank you 

Den

denden4444, 2018-12-28:
@skaarj1989 
Hi David 
Thanks a million for the reply :-)
I'm having issues compiling the client example on esp8266. But will keep trying until I get it going (hence the reason for the other post on bare minumum required)

I have been using links2000 webSockets but ran into problems when trying to have both server and client working on same ESP so I went searching and somehow ended up finding your library and your software for the client.

I have a question on the Arduino forums [https://forum.arduino.cc/index.php?topic=586479.msg3989827#msg3989827](url)
But I have gotten zero replies/help there.
Have also posted similar issues at links200 page on github, and no response for a few weeks now.
I would appreciate it if you could perhaps identify how I replace the links websocket client code with the code used in your library your library.

What I really like about your code is how the connection is easily disconnected from the server since this is exactly what I need to happen for sending only a few bites then closing the connection.

Thank you 

Kind regards 

Den

skaarj1989, 2019-03-23:
Hi,

I have finally managed my library to work with **ESP8266**.
Now you can run both server and client at the same time on one device.

Example code:

```cpp
#include <WebSocketServer.h>
#include <WebSocketClient.h>
using namespace net;

#define _SERIAL Serial

uint16_t port = 81;
WebSocketServer server(port);
WebSocketClient client;

const char *SSID = "SKYNET";
const char *password = "***";

void setup() {
  _SERIAL.begin(115200);
  while (!_SERIAL);

  //_SERIAL.setDebugOutput(true);
  _SERIAL.printf("\nConnecting to %s ", SSID);

  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500); _SERIAL.print(F("."));
  }

  _SERIAL.println(F(" connected"));

  WiFi.printDiag(_SERIAL);

  _SERIAL.print(F("Device IP: ")); _SERIAL.println(WiFi.localIP());

  _SERIAL.print(F("Server running at "));
  _SERIAL.print(WiFi.localIP()); _SERIAL.print(F(":"));
  _SERIAL.println(port);

  // --- Server init:

  server.onConnection([](WebSocket &ws) {
    _SERIAL.print(F("New client: "));
    _SERIAL.println(ws.getRemoteIP());

    ws.onMessage([](WebSocket &ws, WebSocketDataType dataType, const char *message, uint16_t length) {
      _SERIAL.print(F("[server] Received: ")); _SERIAL.println(message);
      ws.send(dataType, message, length);
    });

    ws.onClose([](WebSocket &ws, WebSocketCloseCode code, const char *reason, uint16_t length) {
      _SERIAL.print(ws.getRemoteIP());
      _SERIAL.println(F(" Disconnected"));
    });

    static char message[] = "Hello from server!";
    ws.send(TEXT, message, strlen(message));
  });

  server.begin();

  // --- Client init:

  client.onOpen([](WebSocket &ws) {
    _SERIAL.println(F("[client] Connected"));

    char message[] = "Hello from client!";
    ws.send(TEXT, message, strlen(message));
  });

  client.onMessage([](WebSocket &ws, WebSocketDataType dataType, const char *message, uint16_t length) {
    switch (dataType) {
      case TEXT:
        _SERIAL.print(F("[client] Received: ")); _SERIAL.println(message);
        break;
      case BINARY:
        _SERIAL.println(F("[client] Received binary data"));
        break;
    }

    ws.send(dataType, message, length);
  });

  client.onClose([](WebSocket &ws, const WebSocketCloseCode code, const char *reason, uint16_t length) {
    _SERIAL.println(F("[client] Disconnected\n"));
  });

  client.open("192.168.46.9", 3000);
}

void loop() {
  server.listen(); client.listen();
}
```

 it's not super-fast but works ;)

closed 2019-03-26 by skaarj1989