skaarj1989/mWebSockets #6: webSockets CLient example doesn't compile

denden4444, 2018-12-29:
Hi @skaarj1989 

It seems like the example on the main gitpage here is not compiling , do you possibly have a working example for esp8266 ?
Thank you 
Den 
See below for compile errors:
`
Arduino: 1.8.5 (Windows 8.1), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (512K SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\den\Documents\Arduino\websocketClient_skaarj1989_test_01_181229_1\websocketClient_skaarj1989_test_01_181229_1.ino: In function 'void onOpen(WebSocket&)':

websocketClient_skaarj1989_test_01_181229_1:14: error: no matching function for call to 'WebSocket::send(char [27], size_t)'

   ws.send(message, strlen(message));

                                   ^

C:\Users\den\Documents\Arduino\websocketClient_skaarj1989_test_01_181229_1\websocketClient_skaarj1989_test_01_181229_1.ino:14:35: note: candidates are:

In file included from C:\Users\den\Documents\Arduino\websocketClient_skaarj1989_test_01_181229_1\websocketClient_skaarj1989_test_01_181229_1.ino:5:0:

C:\Users\den\Documents\Arduino\libraries\ArduinoWebSockets\src/WebSocket.h:17:7: note: void WebSocket::send(eWebSocketDataType, const char*, uint16_t)

  void send(const eWebSocketDataType dataType, const char *message, uint16_t length);

       ^

C:\Users\den\Documents\Arduino\libraries\ArduinoWebSockets\src/WebSocket.h:17:7: note:   candidate expects 3 arguments, 2 provided

C:\Users\den\Documents\Arduino\libraries\ArduinoWebSockets\src/WebSocket.h:18:7: note: void WebSocket::send(eWebSocketDataType, const char*, uint16_t, bool)

  void send(const eWebSocketDataType dataType, const char *message, uint16_t length, bool mask);

       ^

C:\Users\den\Documents\Arduino\libraries\ArduinoWebSockets\src/WebSocket.h:18:7: note:   candidate expects 4 arguments, 2 provided

C:\Users\den\Documents\Arduino\websocketClient_skaarj1989_test_01_181229_1\websocketClient_skaarj1989_test_01_181229_1.ino: In function 'void setup()':

websocketClient_skaarj1989_test_01_181229_1:26: error: invalid conversion from 'void (*)(WebSocket&, const char*, uint16_t) {aka void (*)(WebSocket&, const char*, short unsigned int)}' to 'void (*)(WebSocket&, eWebSocketDataType, const char*, uint16_t) {aka void (*)(WebSocket&, eWebSocketDataType, const char*, short unsigned int)}' [-fpermissive]

   client.setOnMessageCallback(onMessage);

                                        ^

In file included from C:\Users\den\Documents\Arduino\websocketClient_skaarj1989_test_01_181229_1\websocketClient_skaarj1989_test_01_181229_1.ino:6:0:

C:\Users\den\Documents\Arduino\libraries\ArduinoWebSockets\src/WebSocketClient.h:18:7: error:   initializing argument 1 of 'void WebSocketClient::setOnMessageCallback(void (*)(WebSocket&, eWebSocketDataType, const char*, uint16_t))' [-fpermissive]

  void setOnMessageCallback(onMessageCallback *callback);

       ^

Multiple libraries were found for "Ethernet.h"
 Used: C:\Users\den\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\Ethernet
 Not used: C:\Users\den\Documents\Arduino\libraries\Ethernet
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
exit status 1
no matching function for call to 'WebSocket::send(char [27], size_t)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

skaarj1989, 2018-12-29:
Hi,

1. Do you have latest version my library?
2. Did you change ``NETWORK_CONTROLLER`` to ``NETWORK_CONTROLLER_WIFI`` in config.h?
For ESP8266 you need to have:
``#define NETWORK_CONTROLLER NETWORK_CONTROLLER_WIFI``
3. In case of errors with arduino-base64 library: Move ``Base64.h`` and ``Base64.cpp`` to ``src`` directory of ArduinoWebSockets library and in Base64.cpp change:
``#include <avr/pgmspace.h>`` to ``#include <pgmspace.h>``

Simple code for ESP8266:

```
#include <WebSocketClient.h>

const char *SSID = "SKYNET";
const char *password = "***";
WebSocketClient client;

void onOpen(WebSocket &ws) {
  Serial.println(F("Connected"));

  char message[] = "Hello from Arduino client!";
  ws.send(TEXT, message, strlen(message));
}

void onClose(WebSocket &ws, const eWebSocketCloseEvent code, const char *reason, uint16_t length) {
  Serial.println(F("Disconnected"));
}

void onMessage(WebSocket &ws, const eWebSocketDataType dataType, const char *message, uint16_t length) {
  switch (dataType) {
    case TEXT:
      Serial.print(F("Received: ")); Serial.println(message);
      break;
    case BINARY:
      Serial.println(F("Received binary data"));
      break;
  }
}

void onError(const eWebSocketError code) {
  Serial.print("Error: "); Serial.println(code);
}

void setup() {
  Serial.begin(57600);
  while (!Serial) ;

  Serial.printf("\nConnecting to %s ", SSID);

  WiFi.begin(SSID, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println(" connected");

  Serial.print(F("Device IP: "));
  Serial.println(WiFi.localIP());
  WiFi.printDiag(Serial);

  // ---

  client.setOnOpenCallback(onOpen);
  client.setOnCloseCallback(onClose);
  client.setOnMessageCallback(onMessage);

  if (!client.open("192.168.46.9", 3000)) {
    Serial.println(F("Connection failed!"));
    while (true) ;
  }
}

void loop() {
  client.listen();
}
```

skaarj1989, 2018-12-30:
Do you have **esp8266** by **ESP8266 Community** version **2.4.1** from Boards Manager installed?  (maybe you could try to reinstall)

I have couple of ESP8266 boards (including Generic and WeMos) and all of them are working with this code.

denden4444, 2018-12-31:
Hi D 

Thanks again for the reply.

Yes 2.4.1 installed and no problems elsewhere. Will try get a fresh install going in a vm to check. 
Dying to get to test your sockets client.

Den

skaarj1989, 2018-12-31:
Try to download new version of ``simple-client.ino`` example. Open project, change value of ``*password`` variable and compile.

denden4444, 2018-12-31:
Hi D 
Ok :-) 
To clarify , the steps are : 
1.Get the Base 64 library and add to arduino libraries as normal
2. Get the crypto library and add to arduino libraries as normal
3. Get latest version of your library and add to arduino libraries as normal
4. Open the upload the latest simple-client .ino and set wifi and password
try to compile and test

Let me know if I have skipped or missed anything, going to try the listed steps and will ping back result.

Thank you kindly 
Den

denden4444, 2019-01-01:
Hi D 
Good news it's working :-) 
Going to perform further testing and will ping back on results if you like  ?
Is there any other way to handle the base 64 and also reduce code space and size ?
What does this do exactly ?
In case of errors with arduino-base64 library: Move Base64.h and Base64.cpp to src directory of ArduinoWebSockets library and in Base64.cpp change:
#include <avr/pgmspace.h> to #include <pgmspace.h>

Kind regards

Den

skaarj1989, 2019-01-01:
Hi,

Base64 and SHA1 algorithms are required. You can read about it in specification or here:
https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API/Writing_WebSocket_servers#Server_Handshake_Response

If you know any better lib for Base64 and/or SHA1 then you can easily add them.
Just modify ``generateSecKey()`` and ``encodeSecKey()`` in ``utility.cpp``

denden4444, 2019-01-02:
Hi D 
Thanks for the reply ...

I was specifically wondering about this line : 
#include <avr/pgmspace.h> to #include <pgmspace.h>
Whats the (subtle) difference ?

Den

skaarj1989, 2019-01-02:
``pgmspace.h`` for ATmega (Boards: Uno, Mega, Pro Mini etc.) and SAMD21 (ARM Cortex M0) is located in ``avr`` while for ESP8266 it's located in root directory.
I use this lib to move strings from dynamic memory to sketch space.

I have started writing this library just because there is no good WebSockets implementation for ATmega (Arduino UNO, Mega etc.) later I decided to support ESP8266 and Cortex M0. 
Because Arduino doesn't have base64 "straight of the box" I had to use an additional lib.

I'll try to use built in base64 for ESP8266 platform.

denden4444, 2019-01-03:
Hi D 

Thanks for the reply.
Just like you I have been looking for the same .. a good library for both webSockets Server and client , the closest i got to it is the one by Links2000 but that is not as flexible in terms of server and client on same library and doesn't handle the connect/disconnect when required easily.
I have your client working with links2000 library now and it seems to be working.
There is a major 1 problem though : 
If the websocket client can't connect to the server it causes a stack dump and the esp resets :-(
If you have some time and keen to give it a test and feedback I would really appreciate it as I am battling to find someone to chat with about the issue and hopefully the solution .. it seems websocket knowledge on these platforms is rather scarce.

I can happily send you the code and setup if you'd like to test.
I was also wondering why your server isn't ESP8266 ready yet ? Would you like some help testing it and getting it going for ESP ?

 Looking forward to your reply.

Kind regards

Den

Puding07, 2019-01-08:
Hi!
Thanks for the fast reply I'm sorry I didn't see you were sending me 2
messages.
I figured out what was the problem: I bought the w5500 ethernet shield and
the company sent me the w5100... tho It's still doesn't work perfectly.
I get stack dump too here are the details:
With your code It gets the ip address and after that it keeps displaying:
"got client" while there is no client connected and when I try to connect
trough the web site it just can't connect it says "Disconnected".
I tried to give static ip address with this code:

IPAddress ip( 192, 168, 0, 177);
IPAddress gateway( 192, 168, 0, 1);
IPAddress subnet( 255, 255, 255, 0);

.
.
.

Ethernet.begin(mac, ip, gateway, gateway, subnet);

and ofc I commented out the if statement for "Can't open ethernet device".
So with this the issue remains.

I'm building up a smart home server and I wanted to stock with the ethernet
shield due to security issues and speed with the wireless esp device.
I would be happy to help you tho. I'm just kinda new to websockets.
If you could help me understand how the websockets works on the arduino I
would really appreciate that cause I'm eagered to get this project done.
I was wondering if I could use my webapp with arduino. The webapp uses
socket.io and it is emitting the messages. Do you know how to get those
emits with the arduino ? I can send you the code for the webapp.
Thank you again for your reply, I hope we can work together,
Aaron

skaarj1989, 2019-04-06:
@denden4444 
In the latest version I've added full support for ESP8266 (server and client)

denden4444, 2019-05-22:
Hi David 

How are you ?

Apologies for the late reply .. I've been overloaded with work :-( 

Could you please clarify a few things please : 
1.  In my post above 

> Re Base64 .. did you know ESP8266 already had Base64 support built in ?

Is there no more need for extra installed library ?

2. The name for your library components would conflict with links2000 webSockets library names if already installed not so ?
Specifically the following : 
        #include <Arduino.h>
        #include <ESP8266WiFi.h>
        #include <ESP8266WiFiMulti.h>
        #include <WebSocketsServer.h>
        #include <WebSocketsClient.h>
        #include <Hash.h>

Is it possible to easily change the names somehow without having to remove the links2000 library ?

I will happily test your library on my ESP's especially with client and server in same ESP, please could you once again provide an example based on latest changes etc. and I will load and test once again. 

Thank you kindly 

Den

haidarshodam, 2019-06-25:
hi @skaarj1989 
i cant compile websocket client example in my arduino uno board with ethernet W5100
cant you help me ?

> Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\defines.h:12:0,

                 from C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\WebSocket.h:4,

                 from C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\WebSocket.cpp:1:

C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\WebSocket.cpp: In member function 'IPAddress net::WebSocket::getRemoteIP()':

C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\platform.h:34:36: error: 'class EthernetClient' has no member named 'remoteIP'

 # define _REMOTE_IP(client) client.remoteIP()

                                    ^

C:\Users\ACER\Documents\Arduino\libraries\mWebSockets-master\src\WebSocket.cpp:112:9: note: in expansion of macro '_REMOTE_IP'

  return _REMOTE_IP(client_);

         ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2021-01-20 by skaarj1989