skaarj1989/mWebSockets #7: Serial monitor freezes after "Initializing..."

Puding07, 2019-01-02:
I get these errors when compiling:

> 

In file included from C:\Users\Aron\Downloads\ArduinoWebSockets-master\examples\simple-server\simple-server.ino:6:0:

C:\Users\Aron\Documents\Arduino\libraries\src/WebSocketClient.h:11:66: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

  bool open(const char *host, uint16_t port = 3000, char path[] = "/");

                                                                  ^

In file included from C:\Users\Aron\Documents\Arduino\libraries\src\WebSocketClient.cpp:1:0:

C:\Users\Aron\Documents\Arduino\libraries\src\WebSocketClient.h:11:66: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

  bool open(const char *host, uint16_t port = 3000, char path[] = "/");

                                                                  ^

C:\Users\Aron\Documents\Arduino\libraries\src\WebSocket.cpp: In member function 'void WebSocket::close(eWebSocketCloseEvent, const char*, uint16_t, bool)':

C:\Users\Aron\Documents\Arduino\libraries\src\WebSocket.cpp:51:15: warning: narrowing conversion of '(int)(((unsigned int)code) >> 8)' from 'int' to 'char' inside { } [-Wnarrowing]

   (code >> 8) & 0xFF,

               ^

C:\Users\Aron\Documents\Arduino\libraries\src\WebSocket.cpp:52:8: warning: narrowing conversion of '(((int)code) & 255)' from 'int' to 'char' inside { } [-Wnarrowing]

   code & 0xFF,

        ^
> 

I'm using arduino uno & W5500 ethernet shield.
I also followed the Requirements and the Installation.
When I open the serial monitor it displays Initializing... but nothing happens after that no matter how much I wait.
Can you please help me ?
Thank you,
Aaron

skaarj1989, 2019-01-02:
Hi,

1. These messages are just warnings not errors.
2. Try to run any example of Ethernet2 library to check if you have correctly connected arduino to W5500
3. Make sure that you have following code in ``config.h``
``
#define NETWORK_CONTROLLER ETHERNET_CONTROLLER_W5500
``
4. Your arduino may hang on initialization because it can't get IP from DHCP. Possible solutions:
a) Add static IP for given MAC address.
b) change this piece of code (in setup function):
```
if (Ethernet.begin(mac) == 0) {
    CONSOLE.println(F("Can't open ethernet device"));
    while (true) ;
}
```
to this:
``
// X is up to you, set these as you like.
Ethernet.begin(mac, IPAddress(192, 168, X, X));
``

Puding07, 2019-01-10:
Hi!
Thanks for the fast reply I already replied to you or I thought so... but I replied to denden4444 😄 
So I figured that the problem was that I bought the w5500 ethernet board but the company sent me the w5100.
After this I edited config.h and the board managed to get an IP address but on the serial monitor I keep getting 'got client' and also I cannot connect to the server via the webapp.

I tried to give static ip address with this code:

IPAddress ip( 192, 168, 0, 177);
IPAddress gateway( 192, 168, 0, 1);
IPAddress subnet( 255, 255, 255, 0);

.
.
.

Ethernet.begin(mac, ip, gateway, gateway, subnet);

and ofc I commented out the if statement for "Can't open ethernet device".
So with this the issue still remains.
If you could help me understand how the websockets works on the arduino I
would really appreciate that cause I'm eagered to get this project done.
I was wondering if I could use my webapp with arduino. The webapp uses
socket.io and it is emitting the messages. Do you know how to get those
emits with the arduino ? I can send you the code for the webapp.
Thank you again for your reply,
Aaron. 👋

skaarj1989, 2019-01-10:
Hi,

Try to use this client code:
https://github.com/skaarj1989/ArduinoWebSockets/blob/master/node.js/chat-client.htm

Send me your webapp client code if you can.

Puding07, 2019-01-15:
I already tried it with the app you sent and ofc I edited in the code the server ip address.
Here is the webapp I am talking about: https://github.com/iamshaunjp/websockets-playlist/tree/lesson-5
Is there a way to handle those events in /public/chat.js ?

skaarj1989, 2019-03-09:
I don't support socket.io for now.

closed 2019-03-09 by skaarj1989