skaarj1989/mWebSockets #8: Get _debugOutput on serial

RenonDis, 2019-01-24:
Hi, thanks for your work on this lib.

I want to debug a failed websocket connection, how can i get _debugOuput calls (in WebSocketClient.cpp for example) to print on serial ?

skaarj1989, 2019-01-24:
Just head to `config.h` and uncomment following lines if you want additional informations in serial monitor:
```
//#define _DEBUG
//#define _DUMP_HANDSHAKE
//#define _DUMP_FRAME_DATA
//#define _DUMP_HEADER
```

RenonDis, 2019-01-24:
Wow thanks for the fast and useful answer. I'm getting nice debug info now.

I'm trying to use WebSocketClient to connect to a python aiohttp server (working with other websocket clients), but I get this error :
```
18:10:36.172 -> Initializing ... 
18:10:40.855 -> Device IP: 134.214.141.162
18:10:40.888 -> [Line #0] HTTP/1.1 101 Switching Protocols
18:10:40.954 -> [Line #1] Upgrade: websocket
18:10:40.987 -> [Line #2] Connection: upgrade
18:10:40.987 -> Error during WebSocket handshake: 'Connection' header value is not 'Upgrade': upgrade
18:10:41.087 -> Connection failed!
```
When commenting out the "upgrade check" in the sources, it works. Any idea why ?

skaarj1989, 2019-01-24:
According to Websocket Protocol specification RFC 6455:
**Upgrade** should be assigned to the **Connection** header.

```
GET /chat HTTP/1.1
Host: server.example.com
Upgrade: websocket
Connection: Upgrade
...
Sec-WebSocket-Version: 13
```
... and you have an **upgrade**

As you can see below I have used **strcmp_P** which is case sensitive, thats why you received an error.

```
if (strcmp_P(value, (PGM_P)F("Upgrade")) != 0) {
...
return false
}
```

If you want you can replace **strcmp_P** with **strcasecmp_P** which is case insensitive.

RenonDis, 2019-01-25:
Thanks a lot for the heads up and the detailed explanation !

I wish you a great day sir.

closed 2019-01-25 by RenonDis