skaarj1989/mWebSockets #11: Arduino UNO net::ERR_CONNECTION_REFUSED

marek307, 2019-05-11:
Hello,

i cant connect to server getting this message:
"Error in connection establishment: net::ERR_CONNECTION_REFUSED"
Im using Arduino UNO.

Thanks for any suggestions.

skaarj1989, 2019-05-12:
Hi,

@marek307 Could you give me more details about your issue?
- WiFi or Ethernet? (which controller)
- What kind of server you are trying to connect? (python, node.js etc.)
- WebSocket or Socket.io?

marek307, 2019-05-12:
Hi,

@skaarj1989 as follow:

- Andurino UNO with Etherneth Shield W5100 (both compatible) as server and ESP8266 D1 mini as client, but server dosent respond to browser clients either
- Server from yours example :)
- WebSocket

marek307, 2019-05-15:
Hello,

i set up server at ESP8266 (similiar code) and in serial monitor popued this:

> [Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.1.190:3000
[Line #2] User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko/20100101 Firefox/66.0
[Line #3] Accept: */*
[Line #4] Accept-Language: pl,en-US;q=0.7,en;q=0.3
[Line #5] Accept-Encoding: gzip, deflate
[Line #6] Sec-WebSocket-Version: 13
[Line #7] Origin: null
[Line #8] Sec-WebSocket-Extensions: permessage-deflate
[Line #9] Sec-WebSocket-Key: QStVN6KV6GFbe4bez0VRUQ==
[Line #10] DNT: 1
[Line #11] Connection: keep-alive, Upgrade

But in client is visable information (without delay):

> ERROR: undefined

Any ideas? :D

skaarj1989, 2019-05-15:
@marek307 I have similar issues. Code that used to work now doesn't and this is frustrating.
I have to investigate this.

marek307, 2019-05-15:
"Security politics"  :D

skaarj1989, 2020-05-30:
Most recent version works with Chrome and Firefox.

closed 2020-05-30 by skaarj1989