skaarj1989/mWebSockets #14: w5500 and esp32 websocket server

MaxESP, 2019-10-01:
dear sir,

I was generic esp32 working with ethernet.h ans spi config

i want to implement websocket server to w5500 with esp32

do you have some idear?

thank a lot

skaarj1989, 2020-06-03:
Hi, sorry for so late reply.
> i want to implement websocket server to w5500 with esp32

It depends if ESP32 supports W5500.

Unfortunately I don't have any ESP32

k-zhai, 2020-06-12:
I'm also interested in knowing if this works for ESP32

MaxESP, 2020-06-13:
For the moment a made w5500 workable just simple http server on esp32. As I am a profesionnal in home automation websocket is very good technology . 
I explain as now all smartphone are very powerfull the fact that you put all html js and css on the phone is easy , than you just have to communicate to the server by websocket to receive command 
make the home automation very fast responsive.   I am have a problem with IOS to implement websocketweb view

skaarj1989, 2020-06-26:
I've received **ESPDUINO-32** and tested it - works fine.

You have to comment ``virtual void begin(uint16_t port=0) =0;`` in **Server.h** located in:
`C:\Users\{your-user-name}\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32`

And during initialization (it may vary for your board):
``Ethernet.init(5);``

MaxESP, 2020-06-26:
I going to try thank a lot

closed 2020-09-21 by skaarj1989