skaarj1989/mWebSockets #17: Large JSON payloads to a websockets client causes arduino to freeze

SamWdW, 2020-07-10:
Thank you for this fantastic project.
I'm running it as a WebSockets client on a Mega 2560 with the W1500 Ethernet Shield.
Having set my buffer size to 4k, and increased the timeout (in config.h), I find that incoming messages with a payload of anything more than about 2000 bytes cause the device to freeze.

(Application: making an OBS Tally-Light, but some OBS scenes send rather long JSON strings if they have a lot of elements in them. Works perfectly with short messages).

Do you have any advice on gracefully handling long incoming messages, even if there's a way of only reading the first portion and scrapping the rest, please?

skaarj1989, 2020-07-10:
Hi,

I've tried to replicate you case. Receiving dummy json (almost 3000 bytes):

`RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2968, MASK=None`
`TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2968, MASK=5dffb57fff91`
`TX BYTES = 2976`
`RX FRAME : OPCODE=8, FIN=True, RSV=0, PAYLOAD-LEN=0, MASK=None`
`Received close frame: code = 0, reason =  `
`TX FRAME : OPCODE=8, FIN=True, RSV=0, PAYLOAD-LEN=2, MASK=581fffbaffd4`
`TX BYTES = 8`

Which version do you use?

SamWdW, 2020-07-10:
Hi David. Many thanks indeed for taking a look.

I'm using the "master" version, as downloaded on 28th June.

I had previously been running with the _DEBUG and _DUMP lines commented out (in case it was a speed thing). Here's the output:

```
Eth Device IP: 192.168.0.104
19:55:28.754 -> [Line #0] HTTP/1.1 101 Switching Protocols
19:55:28.788 -> [Line #1] Connection: Upgrade
19:55:28.822 -> [Line #2] Sec-WebSocket-Accept: 3NskPQR2C8DAHB9Gh+kzvnasBEU=
19:55:28.891 -> [Line #3] Server: WebSocket++/0.8.1
19:55:28.925 -> [Line #4] Upgrade: websocket
19:55:28.960 -> [Line #5] 
19:55:28.960 -> Connected
19:55:28.994 -> Client is Sending: 
19:55:28.994 -> { "request-type": "SetHeartbeat",  "message-id":"123", "enable":true   } 
19:55:29.097 -> TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=73, MASK=446ffeaffc1
19:55:29.166 -> TX BYTES = 79
19:55:29.166 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=67, MASK=None
19:55:30.020 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=563, MASK=None
19:55:32.005 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=552, MASK=None
19:55:33.997 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=551, MASK=None
19:55:35.983 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=550, MASK=None
19:55:38.007 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=561, MASK=None
19:55:40.005 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=561, MASK=None
19:55:41.992 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=550, MASK=None

19:55:44.016 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=563, MASK=None
19:55:45.997 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=562, MASK=None
19:55:46.065 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
19:55:46.304 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
19:55:46.373 -> RX F
```

The "Long Message" appears to be several frames in very quick succession, starting at 19:55:44.016 (NewLine added above for clarity), (prior to that were perfectly processed 2-second "heartbeat" messages coming in which I can confirm can go on for hours without a glitch!). You can see where it then stops mid-diagnostic-output in the last item sent to serial.

Adding up the lengths we get: 563+562+203+206+? = 1534, but I believe what was coming in would be a total of about 3604 bytes (see attached .txt example).
My kBufferMaxSize = 5500;

The compiler reports the following, which seems OK, I think:
Global variables use 6345 bytes (77%) of dynamic memory, leaving 1847 bytes for local variables. Maximum is 8192 bytes.
avrdude: 28336 bytes of flash written

My main loop only contains: `client.listen();`
And at the moment I'm not doing anything with the received message in `client.onMessage([](WebSocket &ws, const WebSocket::DataType &dataType, const char *message, uint16_t length)`    in order to keep things simple at the moment!

Thank you for your guidance,
Sam

[Example JSON from OBS - scene with nested elements.txt](https://github.com/skaarj1989/mWebSockets/files/4904922/Example.JSON.from.OBS.-.scene.with.nested.elements.txt)

skaarj1989, 2020-07-11:
I've just tried to send your example file **1000 times** to Arduino WebSocketClient every **100ms**
No errors.

> Adding up the lengths we get: 563+562+203+206+? = 1534, but I believe what was coming in would be a total of about 3604 bytes

In your output I see many single messages, not one big as you assumed (frames are not fragmented).

Seems like in very last frame Arduino received something weird.

SamWdW, 2020-07-11:
Thank you for trying this out - much appreciated.
I was definitely wrong in my initial assumption that it was "one large message", as the debug output now shows; although I'm confused when trying to match the payload sizes to the txt captured (using a browser running a WebSocket client being sent the same messages).

By "something weird" do you mean something like a byte in the incoming string with a NULL value, which I can't see? 

Alternatively, if it's not a size problem, is it a speed issue instead, perhaps? I suspect the incoming messages may be more tightly spaced than 100mS when several come in at once - could that be the cause of the problem I'm seeing, rather than the size of the payload?

19:55:45.997 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=562, MASK=None
19:55:46.065 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
19:55:46.304 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
19:55:46.373 -> RX F

Gaps between these above are:
68ms, 239ms, 69ms

I've tried with another set of incoming messages, and get the following before a similar freeze occurs:
09:17:30.108 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=564, MASK=None
09:17:31.269 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=175, MASK=None
09:17:31.336 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=804, MASK=None
09:17:31.406 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=821, MASK=None
09:17:31.576 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=178, MASK=None
09:17:31.645 -> RX FR⸮

and again, to see how repeatable it is:

The common factor is a gap of 67-70 mS between frames.
Presumably there's a relationship between buffer size and minimum gap between frames?
At the moment I'm not doing anything else in the sketch, so there's no time-consuming code in the onMessage, to rule out delays there.

Thank you for taking a look!

SamWdW, 2020-07-11:
Also, I note that I get the same behaviour (i.e. freezing with rapid incoming messages) when the _DUMP_HEADER is switched off, and replaced by printing a single character in the onMessage - so it wouldn't appear to be a speed penalty caused by the diagnostic output being on.

skaarj1989, 2020-07-11:
I tried to reduce interval to **50ms** - still no error.
Can you capture last couple of frames and attach them here?

SamWdW, 2020-07-11:
That's interesting... here are the last frames, captured with _DUMP_FRAME_DATA switched on:

```
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=555, MASK=None
09:46:18.197 -> {
09:46:18.197 ->     "current-profile": "Untitled",
09:46:18.264 ->     "current-scene": "TALLY:1:OFF",
09:46:18.297 ->     "pulse": true,
09:46:18.297 ->     "recording": false,
09:46:18.331 ->     "recording-paused": false,
09:46:18.365 ->     "stats": {
09:46:18.365 ->         "average-frame-time":RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
09:46:19.695 -> {
09:46:19.695 ->     "duration": 300,
09:46:19.695 ->     "from-scene": "TALLY:1:OFF",
09:46:19.729 ->     "name": "Fade",
09:46:19.764 ->     "to-scene": "## Video link (for TX) ###############",
09:46:19.831 ->     "type": "fade_transition",
09:46:19.866 ->     "update-type": "TransiRX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
09:46:20.001 -> {
09:46:20.001 ->     "duration": 300,
09:46:20.035 ->     "from-scene": "TALLY:1:OFF",
09:46:20.069 ->     "name": "Fade",
09:46:20.103 ->     "to-scene": "## Video link (for TX) ###############",
09:46:20.139 ->     "type": "fade_transition",
09:46:20.173 ->     "update-type": "TransiRX F
```

and another attempt, from start-up:

```
Eth Device IP: 192.168.0.104
09:47:50.517 -> [Line #0] HTTP/1.1 101 Switching Protocols
09:47:50.552 -> [Line #1] Connection: Upgrade
09:47:50.585 -> [Line #2] Sec-WebSocket-Accept: KHAQjxAW2qEoZVo6J+zpyNYmSoo=
09:47:50.655 -> [Line #3] Server: WebSocket++/0.8.1
09:47:50.689 -> [Line #4] Upgrade: websocket
09:47:50.723 -> [Line #5] 
09:47:50.723 -> Connected
09:47:50.758 -> Client is Sending: 
09:47:50.758 -> { "request-type": "SetHeartbeat",  "message-id":"123", "enable":true   } 
09:47:50.861 -> TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=73, MASK=ffb8ff884942
09:47:50.929 -> { "request-type": "SetHeartbeat",  "message-id":"123", "enable":true   } 
09:47:50.999 -> TX BYTES = 79
09:47:50.999 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=67, MASK=None
09:47:51.100 -> {
09:47:51.100 ->     "enable": true,
09:47:51.100 ->     "message-id": "123",
09:47:51.134 ->     "status": "ok"
09:47:51.134 -> }
09:47:51.922 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
09:47:51.990 -> {
09:47:51.990 ->     "duration": 300,
09:47:52.024 ->     "from-scene": "TALLY:1:OFF",
09:47:52.057 ->     "name": "Fade",
09:47:52.091 ->     "to-scene": "## Video link (for TX) ###############",
09:47:52.126 ->     "type": "fade_transition",
09:47:52.160 ->     "update-type": "TransiRX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=583, MASK=None
09:47:52.264 -> {
09:47:52.264 ->     "current-profile": "Untitled",
09:47:52.298 ->     "current-scene": "## Video link (for TX) ###############",
09:47:52.366 ->     "pulse": false,
09:47:52.400 ->     "recording": false,
09:47:52.433 ->     "recording-paused": false,
09:47:52.467 ->     "stats": {
09:47:52.467 ->  RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
09:47:52.536 -> {
09:47:52.536 ->     "duration": 300,
09:47:52.570 ->     "from-scene": "TALLY:1:OFF",
09:47:52.604 ->     "name": "Fade",
09:47:52.604 ->     "to-scene": "## Video link (for TX) ###############",
09:47:52.673 ->     "type": "fade_transition",
09:47:52.707 ->     "update-type": "TransiRX F
```

The plain-text of what's being sent is in the .txt file, which appears not to be anything odd!

Looking for other differences between what you're seeing and what I'm seeing...
I'm using the library ethernet.h (v2.00, I believe) on a W5100

I also tried reducing the buffer size, as incoming messages seem to be lower than I'd originally thought, so tried it at 1000, and got this:

```
09:55:02.358 -> TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=73, MASK=ffbaffda1e68
09:55:02.424 -> TX BYTES = 79
09:55:02.424 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=554, MASK=None
09:55:02.492 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=67, MASK=None
09:55:04.135 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=566, MASK=None
09:55:06.145 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=565, MASK=None
09:55:08.132 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=565, MASK=None
09:55:08.645 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
09:55:08.954 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
09:55:09.022 -> Unsupported frame size = 3023
09:55:09.057 -> TX FRAME : OPCODE=8, FIN=True, RSV=0, PAYLOAD-LEN=2, MASK=74ffa260ffea
09:55:09.126 -> TX BYTES = 8
09:55:10.055 -> Disconnected - connection closed
```

Which is repeatable. The "Unsupported frame size" goes away when I increase kBufferMaxSize again (this time to 6000), but the freeze issue then comes back:

```
10:00:13.775 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=67, MASK=None
10:00:14.153 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=565, MASK=None
10:00:16.153 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=566, MASK=None
10:00:16.761 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=203, MASK=None
10:00:17.069 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=206, MASK=None
10:00:17.136 -> RX F
```

Thank you for your help.

skaarj1989, 2020-07-11:
It would be helpful to see that frame of size **3023**. Maybe you could capture it in browser or node.

SamWdW, 2020-07-11:
Here's a "large frame" captured using the SimpleWebSockets browser extension in Chrome (please see .txt file).
Interestingly, with a large buffer size set for kBufferMaxSize the RX FRAME debug output shows small-ish payloads, but many of them (until it freezes), as you noted previously. When a large buffer size is set the _same_ incoming data gives the "Unsupported frame size" error, suggesting that the frames seem to be getting merged when read, perhaps?

I'm using OBS with the Websockets plugin as the server to transmit these. Short messages work fine every time!

OBS: https://obsproject.com/
OBS Websockets server plugin: https://github.com/Palakis/obs-websocket 

[Websockets from OBS captured in Chrome WebsocketClient.txt](https://github.com/skaarj1989/mWebSockets/files/4906542/Websockets.from.OBS.captured.in.Chrome.WebsocketClient.txt)

Is there a more helpful way of capturing them?
Thank you!

skaarj1989, 2020-07-11:
> suggesting that the frames seem to be getting merged when read, perhaps?

Message is merged in only one case: fragmented frames (if **FIN** is equal to False)

SamWdW, 2020-07-11:
My guess is that the server isn't usually talking to tiny devices!

Do you think this is a running-out-of-memory issue? I don't seem to be completely out of dynamic memory, even with a kBufferMaxSize of 5000 or 6000 bytes.
Thinking out loud: if so could an interesting feature be to add a "discard whitespace" at the lowest-level of receiving, or some other way of handling this combination of size/speed more gracefully please?

skaarj1989, 2020-07-12:
Hi,
I've just made some fixes in [dev](https://github.com/skaarj1989/mWebSockets/tree/dev) branch. It could help.

SamWdW, 2020-07-13:
You are a genius, sir!
I am delighted to confirm that it now works perfectly with the messages I was throwing at it that caused challenges before. This is excellent, thank you.

Working output (LEN fields are interesting):
```
09:43:10.777 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=580, MASK=None
09:43:12.779 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=581, MASK=None
09:43:14.750 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=581, MASK=None
09:43:14.919 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:15.263 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:15.331 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=3006, MASK=None
09:43:15.467 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=168, MASK=None
09:43:16.769 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=579, MASK=None
09:43:17.211 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:17.517 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:17.585 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2174, MASK=None
09:43:17.687 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=160, MASK=None
09:43:18.748 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=582, MASK=None
09:43:19.332 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:19.639 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:19.707 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=3006, MASK=None
09:43:19.844 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=168, MASK=None
09:43:20.773 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=587, MASK=None
09:43:22.692 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:22.761 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=582, MASK=None
09:43:23.000 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:23.068 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2174, MASK=None
09:43:23.172 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=160, MASK=None
09:43:24.305 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:24.646 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:24.713 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=3006, MASK=None
09:43:24.849 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=168, MASK=None
09:43:24.917 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=586, MASK=None
09:43:26.551 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=222, MASK=None
09:43:26.755 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=571, MASK=None
09:43:26.891 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=225, MASK=None
09:43:26.959 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2174, MASK=None
09:43:27.063 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=160, MASK=None
09:43:28.778 -> RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=581, MASK=None
```

More cycle-testing to follow, but looking good - much appreciated!

closed 2020-07-16 by SamWdW