skaarj1989/mWebSockets #21: Base64 library

gitneko, 2020-09-21:
Currently I'm looking into this library for a project of mine (part of #19) and I've noticed that this library requires a base64 implementation, but none of the ones I've been looking at in the Arduino Library Manager actually define the needed functions.

Which library do you need installed to have these base64 functions around? Would you be willed to migrate to a different library, if the library can not be found in the library manager?

Currently I have the following one installed: <https://github.com/agdl/Base64>

And the following two seem sane from a quick view:
- <https://github.com/Densaugeo/base64_arduino>
- <https://github.com/dojyorin/arduino_base64>

skaarj1989, 2020-09-21:
It's bundled.

gitneko, 2020-09-21:
Ah, yes it's bundled with git, but not in the latest version. Therefore this was getting overlooked by me and the compiler... :) 

Can I assume you will make a new release once this library works fine on SAMD (after my feedback)?

skaarj1989, 2021-01-20:
both **base64** and **CryptoLegacy** are bundled in newest release.

closed 2021-01-20 by skaarj1989