skaarj1989/mWebSockets #24: W5100 Server not upgrade connection

tltart, 2020-12-08:
Hello,
Your library helps me out a lot. However, I can't establish a connection.

Here's what it gives out DEBUG.

And  ws.send(message, length);  the first parameter is missing...
I used ws.send( {}, message, length);

Server running at 192.168.88.116:88
[Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.88.116:88
[Line #2] Connection: keep-alive
[Line #3] Cache-Control: max-age=0
[Line #4] Upgrade-Insecure-Requests: 1
[Line #5] User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
[Line #6] Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
[Line #7] Accept-Encoding: gzip, deflate
[Line #8] Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7
[Line #9]

skaarj1989, 2020-12-08:
Hi,

Which board, shield (ehternet module) and library do you use? Is your server **simple-server.ino** or something else?

Here is my log for `ws.send({}, message, length);` using **Mega2560** with **W5100** and stock Ethernet library.
```
[Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.46.180:3000
[Line #2] Connection: Upgrade
[Line #3] Pragma: no-cache
[Line #4] Cache-Control: no-cache
[Line #5] User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
[Line #6] Upgrade: websocket
[Line #7] Origin: file://
[Line #8] Sec-WebSocket-Version: 13
[Line #9] Accept-Encoding: gzip, deflate
[Line #10] Accept-Language: pl-PL,pl;q=0.9,en-US;q=0.8,en;q=0.7
[Line #11] Sec-WebSocket-Key: 1iAhzShA8oWoG/ofvM+tYg==
[Line #12] Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
[Line #13] 
New client: 192.168.46.31
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=15, MASK=27ffd2ffb8ffa0
[object Object]
Received: [object Object]
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=15, MASK=None
[object Object]
TX BYTES = 17
```

tltart, 2020-12-08:
Я тоже использую MEGA 2560 и w5100. Запустил simple-server.ino ... тот же результат. Это может быть связано с F строками?

Initializing ...
Server running at 192.168.88.116:88
[Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.88.116:88     
[Line #2] Connection: keep-alive      
[Line #3] Upgrade-Insecure-Requests: 1
[Line #4] User-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 13_2_3 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.0.3 Mobile/15E148 Safari/604.1
[Line #5] Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
[Line #6] Accept-Encoding: gzip, deflate
[Line #7] Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7
[Line #8]

skaarj1989, 2020-12-08:
Does your server work with any desktop browser? There is no `Upgrade` header in your iPhone client request

tltart, 2020-12-08:
The server is not accessible from the browser

skaarj1989, 2020-12-08:
Server rejects client because there is no `Upgrade` header.

tltart, 2020-12-08:
import websocket

def on_message(ws, message):
    print(message)
def create_connection():
    websocket.enableTrace(True)
    ws = websocket.WebSocketApp("ws://192.168.88.116:88", on_message=on_message)
    ws.run_forever()

if __name__ == '__main__':
    create_connection()

skaarj1989, 2020-12-08:
Try this:
```py
from websocket import create_connection
ws = create_connection("ws://192.168.88.116:88")
print("Sending 'Hello, World'...")
ws.send("Hello, World")
print("Sent")
print("Receiving...")
result =  ws.recv()
print("Received '%s'" % result)
ws.close()
```

tltart, 2020-12-08:
Arduino doesn't respond.

All the same....

In sketch  wss, on client ws. I have tried and so and so....

skaarj1989, 2020-12-08:
And what about connecting via javascript?

tltart, 2020-12-08:
I have a complex project structure, it is the Python script that controls the drone via websocket

skaarj1989, 2020-12-08:
Ok, but first we have to find out the issue. Just try simple javascript code.

tltart, 2020-12-08:
Ok

skaarj1989, 2020-12-08:
I have foud very weird behaviour.
Open `platform.h` and change **MAX_SOCK_NUM** to number <= 8;
```cpp
constexpr uint8_t kMaxConnections{ MAX_SOCK_NUM };
```

tltart, 2020-12-08:
constexpr uint8_t kMaxConnections {MAX_SOCK_NUM};
#elif NETWORK_CONTROLLER == ETHERNET_CONTROLLER_ENC28J60
#  include <IPAddress.h>
#  include <UIPEthernet.h>
constexpr uint8_t kMaxConnections = 8; 

and

constexpr uint8_t kMaxConnections = MAX_SOCK_NUM;
#elif NETWORK_CONTROLLER == ETHERNET_CONTROLLER_ENC28J60
#  include <IPAddress.h>
#  include <UIPEthernet.h>
constexpr uint8_t kMaxConnections = 8;

not working.....

skaarj1989, 2020-12-08:
I mean replace **MAX_SOCK_NUM** with number

tltart, 2020-12-08:
Understood...

But that doesn't work either

skaarj1989, 2020-12-08:
Are you sure that you removed **MAX_SOCK_NUM**? from header?

tltart, 2020-12-08:
I don't quite understand where to delete it...

I did  in platform.h
```
#if NETWORK_CONTROLLER == ETHERNET_CONTROLLER_W5X00
#  include <Ethernet.h>
/** Specifies maximum number of clients connected to server. */
constexpr uint8_t kMaxConnections = 8;

```

Need to change something else?

skaarj1989, 2020-12-08:
If it did not work then I need detailed specs of your board and ethernet module to reproduce your issue.

tltart, 2020-12-08:
```
MEGA 2560 PRO Embed CH340G
board = megaatmega2560
```
ethernet module

```
Mini W5100 LAN Ethernet Shield

````

skaarj1989, 2020-12-08:
I have two different **Mega2560** and couple of **W5100** modules, everything works fine on all of them.

tltart, 2020-12-08:
The fact is that the web server works without problems. Here, the server doesn't add a client.
I ran out of thoughts(((

skaarj1989, 2020-12-08:
Do you have any other board to test?
Check if you can run **simple-client.ino** without problems.

tltart, 2020-12-08:
Tomorrow I will try....

Thank you for your time

tltart, 2020-12-10:
Hello 0_o ))
I continue to understand.
Found a function that is not running...

String    " client.println(F("HTTP/1.1 101 Switching Protocols"));"   not running...
After board is restart.....

Any ideas?

```
void WebSocketServer::_acceptRequest(NetClient &client, const char *secKey) {
  char acceptKey[28]{};
  encodeSecKey(acceptKey, secKey);
  char secWebSocketAccept[50]{};
  strcpy_P(secWebSocketAccept, (PGM_P)F("Sec-WebSocket-Accept: "));
  strcat(secWebSocketAccept, acceptKey);

  client.println(F("HTTP/1.1 101 Switching Protocols"));

  client.println(F("Upgrade: websocket"));

  client.println(F("Connection: Upgrade"));
  client.println(secWebSocketAccept);
  client.println();
}
```

skaarj1989, 2020-12-10:
It'll be hard since I can't reproduce your issue. Please try to upload `simple-client.ino` so we could verify if the bug is inside WebSocketServer or WebSocket class.

tltart, 2020-12-10:
Client is work

skaarj1989, 2020-12-10:
Try: `client.open("echo.websocket.org", 80);`

tltart, 2020-12-10:
```
client.open("echo.websocket.org", 80);
```
```
Device IP: 192.168.88.208
Connection failed!
````
I use master ...

skaarj1989, 2020-12-10:
Uncomment these lines in `config.h`
```cpp
//#define _DEBUG
//#define _DUMP_HANDSHAKE
//#define _DUMP_HEADER
//#define _DUMP_FRAME_DATA
```

tltart, 2020-12-10:
```
Initializing ... 
Device IP: 192.168.88.208
Error in connection establishment: net::ERR_CONNECTION_REFUSED
Connection failed!
```

tltart, 2020-12-10:
Something is starting to work.... see

skaarj1989, 2020-12-10:
You're welcome, I appreciate every feedback.

closed 2020-12-11 by tltart