skaarj1989/mWebSockets #25: Mega2560 Websocket server. Garbage in the channel TCP.

tltart, 2020-12-11:
Hello.

Along with useful data comes some garbage. What could it be?
Some kind of counter inside the library...
```
payloadString= pult=@
var= pult
val= @1
60
1
60
1
59
1
58
1
57
1
56
1
55
1
54
1
53
1
52
1
51
1
50
1
49
payloadString= pult=@
var= pult
val= @1
48
1
48
1
```

skaarj1989, 2020-12-11:
First uncomment all debug macros, I need informations like that:
```
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=15, MASK=27ffd2ffb8ffa0
[object Object]
Received: [object Object]
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=15, MASK=None
```
Arduino server code is also welcomed.

tltart, 2020-12-11:
simple-server.ino

I send to server once "w" and after once "@"

```
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=72fffefff116
1
6
1
5
1
4
1
3
1
2
1
1
pult=w
Received: pult=w
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=w
TX BYTES = 8
1
12
1
12
1
11
1
10
1
9
1
8
1
7
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=ffab3ffe0ffce
1
6
1
5
1
4
1
3
1
2
1
1
pult=w
Received: pult=w
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=w
TX BYTES = 8
1
24
1
24
1
23
1
22
1
21
1
20
1
19
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=20fff1ffc7fffc
1
18
1
17
1
16
1
15
1
14
1
13
pult=@
Received: pult=@
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=@
TX BYTES = 8
1
12
1
12
1
11
1
10
1
9
1
8
1
7
RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=ff95ffc7ffdcff86
1
6
1
5
1
4
1
3
1
2
1
1
pult=@
Received: pult=@
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=@
TX BYTES = 8
```

skaarj1989, 2020-12-11:
Can you send client code? I'll try to reproduce your issue.

tltart, 2020-12-11:
```
import keyboard
from tkinter import *
import threading
import websocket
import time

buff = []

def print_pressed_keys(e):
    print(e, e.event_type, e.name)

def window():
    window = Tk()
    window.title("......................")
    lbl = Label(window, text="Catch kyeboard")
    lbl['bg'] = "red"
    lbl.grid(column=0, row=0)
    window.attributes("-topmost", True)
    window.mainloop()

def on_message(ws, message):
    print(message)

def on_open(ws):

    keyboard.hook(lambda x: new_foo(ws, x.name))

def new_foo(ws, x):
    while keyboard.is_pressed('w'):
            ws.send("pult=" + x)
            time.sleep(0.1)

    ws.send("pult=@")

def create_connection():
    websocket.enableTrace(True)
    ws = websocket.WebSocketApp("ws://192.168.88.116:3000", on_message=on_message, on_open=lambda a: on_open(ws))
    ws.run_forever()

thread_1 = threading.Thread(target=window, name="tkinter", daemon=True)
thread_2 = threading.Thread(target=create_connection, name="connection", daemon=True)

thread_1.start()
thread_2.start()
thread_1.join()
```

skaarj1989, 2020-12-11:
I managed to run your code, everything works fine, no garbage data:
```
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=26, MASK=None
Hello from Arduino server!
TX BYTES = 28

RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=ff862c65ffe2
pult=w
Received: pult=w

TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=w
TX BYTES = 8

RX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=ffa3ffc5ffed43
pult=@
Received: pult=@

TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=6, MASK=None
pult=@
TX BYTES = 8
```
I think that something may be wrong with your ethernet module.
Once I had problems with garbage data, after about an hour of debugging I found that my shield was not quite right attached to arduino (some pins were bent)

tltart, 2020-12-11:
Yes. Most...

skaarj1989, 2020-12-11:
I think that this does not comes from internal library code. Try to comment `_SERIAL.println` inside `onMessage` callback

tltart, 2020-12-11:
It is strange that the data is transmitted for a very long time. They are waiting in line. If I pass "w" many times in a row (just by holding down the button), the data will go on for a long time, although I released the "w" button a long time ago.

closed 2020-12-14 by tltart