skaarj1989/mWebSockets #26: Wrong value in WebSocketServer.cpp cause a stack overflow error

Ozakest, 2020-12-17:
Hi
I'm currently works whith **VSode & PlatformIo**
I use this line lib_deps = **skaarj1989/mWebSockets @ ^1.4.0** to install mWebSockets libray
my code compil perfect but i have somme trouble in execution when the client connects.

in WebSocketServer.cpp
```

void WebSocketServer::_acceptRequest(NetClient &client, const char *secKey) {
  char acceptKey[28]{};
  encodeSecKey(acceptKey, secKey);

  char secWebSocketAccept[50]{};
  strcpy_P(secWebSocketAccept, (PGM_P)F("Sec-WebSocket-Accept: "));
  strcat(secWebSocketAccept, acceptKey);

  client.println(F("HTTP/1.1 101 Switching Protocols"));
  //client.println(F("Server: Arduino"));
  //client.println(F("X-Powered-By: mWebSockets"));
  client.println(F("Upgrade: websocket"));
  client.println(F("Connection: Upgrade"));
  client.println(secWebSocketAccept);
  client.println();
}
```

cause a stack overflow error

I found a solution

char acceptKey[28]{}; muste to be **char acceptKey[29]{};**
and   
char secWebSocketAccept[50]{}; muste to be **char secWebSocketAccept[51]{}**

// 22 characters for header + 28 for accept key + 1 for NULL

it's same value of 1.33 library
best regard

skaarj1989, 2020-12-17:
Hi, thanks for info, in **master** everything is fine.
Can you tell me where did you find lib with error?

Ozakest, 2020-12-17:
the errors are on line 381 & 384

skaarj1989, 2020-12-17:
I mean: from where did you download that lib with error. I looked into official 1.3.3 release and there is no error.

Ozakest, 2020-12-17:
I installed the library with platformIO by following the following recommendation

Library Dependencies platformio.ini

  # RECOMMENDED
  # Accept new functionality in a backwards compatible manner and patches
  **skaarj1989/mWebSockets @ ^1.4.0**

skaarj1989, 2020-12-18:
Thanks for pointing this issue, looks like it's been on platformio for quite a long time. I increment version to 1.4.1 so hopefuly a crawler will notice it.

skaarj1989, 2021-01-20:
New version available on platformio.

closed 2021-01-20 by skaarj1989