skaarj1989/mWebSockets #27: error 1006

gfduck, 2021-01-18:
I use example of simple client websocket, in localhost works correctly  but in my hosting appear error 1006.
The Arduino connect and desconnect after 1 second in Server.

Can you help me?

This is my code:

#include <WebSocketClient.h>
using namespace net;

byte mac[]{ 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
# define _SERIAL Serial

WebSocketClient client;

void setup() {
  _SERIAL.begin(9600);

  Ethernet.begin(mac); //, ip);

  _SERIAL.print(F("Device IP: "));
  _SERIAL.println(Ethernet.localIP());

  client.onOpen([](WebSocket &ws) {
    _SERIAL.println(F("Connected"));

    constexpr char message[]{ "Hello from Arduino client!" };
    ws.send(WebSocket::DataType::TEXT, message, strlen(message));
  });

  client.onMessage([](WebSocket &ws, const WebSocket::DataType &dataType,
                     const char *message, uint16_t length) {
    switch (dataType) {
    case WebSocket::DataType::TEXT:
      _SERIAL.println(F("Received text message"));
      _SERIAL.println(message);
      break;
    case WebSocket::DataType::BINARY:
      _SERIAL.println(F("Received binary data"));
      break;
    }

    ws.send(dataType, message, length); // echo back to server
  });

  client.onClose(
    [](WebSocket &ws, const WebSocket::CloseCode &code, const char *reason,
      uint16_t length) { _SERIAL.println(F("Disconnected")); });

  if (!client.open("server.com", 80, "/chat")) {
    _SERIAL.println(F("Connection failed!"));
    while (true)
      ;
  }
}

void loop() { client.listen(); }

[image]

skaarj1989, 2021-01-19:
Hi @gfduck 
Could you post arduino log?

gfduck, 2021-01-19:
Thanks for answer!

This is a log from Arduino. I use W5100 ethernet.

In localhost works ok but in hosting not work.

[image]

skaarj1989, 2021-01-19:
Is your server *ws* or *wss*?
Please enable debug macros in `config.h`

gfduck, 2021-01-19:
Thanks for answer!.

This is file config.h, how to enable enable debug macros?

Can you send the correct content of file?

Thanks!

#pragma once

/** @file */

/**
 * @def _DEBUG Enables __debugOutput function.
 * @def _DUMP_HANDSHAKE Prints any handshake request/response on Serial output.
 * @def _DUMP_HEADER Prints frame header on Serial output.
 * @def _DUMP_FRAME_DATA Prints frame data on Serial output.
*/

/**
 * @def NETWORK_CONTROLLER
 * @brief Specifies network controller, available values:
 *  - ETHERNET_CONTROLLER_W5X00
 *  - ETHERNET_CONTROLLER_ENC28J60
 *  - NETWORK_CONTROLLER_WIFI
 */

//#define _DEBUG
//#define _DUMP_HANDSHAKE
//#define _DUMP_HEADER
//#define _DUMP_FRAME_DATA

#define NETWORK_CONTROLLER ETHERNET_CONTROLLER_W5X00

/** Maximum size of data buffer - frame payload (in bytes). */
constexpr uint16_t kBufferMaxSize{ 256 };
/** Maximum time to wait for endpoint response (in milliseconds). */
constexpr uint16_t kTimeoutInterval{ 5000 };

gfduck, 2021-01-19:
My server is ws

skaarj1989, 2021-01-19:
Just uncomment these four defines. Are you trying to connect by passing ip address or http://?

gfduck, 2021-01-19:
I try to connect with url.

This is a log debug:
[image]

skaarj1989, 2021-01-19:
Ok, now it's clear.
Go into `WebSocketClient.cpp` and replace `strcmp_P` with `strcasecmp_P` in the line 172.

gfduck, 2021-01-19:
ohhh it`s awesome!!

it`s working now!!

Thanks a lot!.

Other problem,  In few time to connect arduino is disconnect. 
This is a log.

I  attach screenshot.
[image]

gfduck, 2021-01-19:
In webserver is error 1002

skaarj1989, 2021-01-19:
Any debug message?

gfduck, 2021-01-19:
During this days I will do some test on the library and I will let you know! I am so thankful for your help

skaarj1989, 2021-01-19:
You're welcome.

closed 2021-06-16 by skaarj1989