skaarj1989/mWebSockets #28: Server running at 0.0.0.0:3000

JumpMan35, 2021-01-20:
Hello,
I'am using Ethernet Shield (5100) + Arduino Uno
I have followed the procedure, but the debug console is still raiming the same  
```
Initializing ...
Server running at 0.0.0.0:3000
```
Let me know how to debug this issue, I took to much time to figure out but I can't solve it.
Thanks

Please find my attached code based on simple server

```
#include <WebSocketServer.h>
#include <Ethernet.h>
using namespace net;

#define _SERIAL Serial
#define PinForEthernet 10
#define _DEBUG
#define _DUMP_HANDSHAKE
#define _DUMP_FRAME_DATA
#define _DUMP_HEADER
#define NETWORK_CONTROLLER ETHERNET_CONTROLLER_W5X00

#if NETWORK_CONTROLLER == NETWORK_CONTROLLER_WIFI
constexpr char kSSID[]{ "SKYNET" };
constexpr char kPassword[]{ "***" };
#else
byte mac[]{ 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 1, 160);
#endif

constexpr uint16_t port = 3000;
WebSocketServer wss(port);

void setup() {
  
  _SERIAL.begin(9600);
  while (!_SERIAL)
    ;

#if NETWORK_CONTROLLER == NETWORK_CONTROLLER_WIFI
  //_SERIAL.setDebugOutput(true);
  _SERIAL.printf("\nConnecting to %s ", kSSID);

  WiFi.mode(WIFI_STA);
  WiFi.begin(kSSID, kPassword);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    _SERIAL.print(F("."));
  }

  _SERIAL.println(F(" connected"));

  WiFi.printDiag(_SERIAL);

  _SERIAL.print(F("Device IP: "));
  _SERIAL.println(WiFi.localIP());
#else
  _SERIAL.println(F("Initializing ... "));

#if NETWORK_CONTROLLER == ETHERNET_CONTROLLER_W5X00
   Ethernet.init(53);
#endif

  Ethernet.begin(mac, ip);

  _SERIAL.print(F("Server running at "));
  _SERIAL.print(Ethernet.localIP());
  _SERIAL.print(F(":"));
  _SERIAL.println(port);
#endif

  wss.onConnection([](WebSocket &ws) {
    ws.onMessage([](WebSocket &ws, const WebSocket::DataType &dataType,
                   const char *message, uint16_t length) {
      switch (dataType) {
      case WebSocket::DataType::TEXT:
        _SERIAL.print(F("Received: "));
        _SERIAL.println(message);
        break;
      case WebSocket::DataType::BINARY:
        _SERIAL.println(F("Received binary data"));
        break;
      }

      ws.send(dataType, message, length);
    });

    ws.onClose(
      [](WebSocket &ws, const WebSocket::CloseCode &code, const char *reason,
        uint16_t length) { _SERIAL.println(F("Disconnected")); });

    _SERIAL.print(F("New client: "));
    _SERIAL.println(ws.getRemoteIP());

    const char message[]{ "Hello from Arduino server!" };
    ws.send(WebSocket::DataType::TEXT, message, strlen(message));
  });

  wss.begin();
}

uint32_t previousTime = 0;

void loop() { wss.listen(); }
```

skaarj1989, 2021-01-20:
First make sure that you are able to create server with Ethernet library (WebServer example)

JumpMan35, 2021-01-20:
Thank for your reply.
Previously, I have done a simple web server with Ethernet library.
I have do again today, It' s working well with Ethernet library

skaarj1989, 2021-01-20:
Update this line `Ethernet.init(53);` to proper pin value.

JumpMan35, 2021-01-20:
You 're right, I've refreshed with `Ethernet.init(10);` in my case Arduino Ethernet Shield
But it still the same issue with WebServerSocket code `Server running at 0.0.0.0:3000`  and the WebServer is working

skaarj1989, 2021-01-20:
Your issue is not related to **mWebSockets** library. Ip address is fetched from `Ethernet.localIP()`, something is wrong with your setup (mac or ip maybe?) or wiring.

JumpMan35, 2021-01-20:
In the basic web server, I've taken the same MAC & IP adresses and using `Ethernet.localIP` function as in the WebSocketServer example . No problem.
Concerning the wiring, I didn't change anything since I've got the Arduino Uno / Ethernet Shield W5100
Do you have an another idea to investigate with WebSocketServer library or debug mode  ?

skaarj1989, 2021-01-20:
You could try `Ethernet.begin(mac);` instead of `Ethernet.begin(mac, ip);`

JumpMan35, 2021-01-20:
Thank you for help
Good news, it's progressing well
```
Server running at 192.168.1.101:3000
New client: 192.168.1.102
```
You know how to fix the IP adress ?

skaarj1989, 2021-01-20:
Don't know for sure but it may be related to ip address leasing in router.

closed 2021-02-01 by skaarj1989