skaarj1989/mWebSockets #29: server closes the websocket after 5 seconds

stevstrong, 2021-02-10:
HI,
I am using the simple server example from your repo for STM32 architecture and W5500 module combined with the stock Ethernet library.
I have the problem that after 5 seconds after I start the client application (Firefox on my PC) the client reports that the websocket is closed by the server (this example).

Digging into the code I arrived to WebSocketServer.cpp:
https://github.com/skaarj1989/mWebSockets/blob/cdc772b52ad008084146310388352a73d68332bf/src/WebSocketServer.cpp#L70-L91
In particular, it seems that line 91 should be only called in the else case of line 74 condition. Otherwise, as it currently is, for the first time when ws is NULL, in line 91, ws will be non-null in line 80 but there are no available data to read, so that the websocket `_read()` function ([see here](https://github.com/skaarj1989/mWebSockets/blob/cdc772b52ad008084146310388352a73d68332bf/src/WebSocket.cpp#L230-L239)) called by `_readFrame` will timeout exactly after 5 seconds.

Changing the above mentioned code line in the following way solves the problem:
```
    if (!ws) {
      const auto end = &m_sockets[kMaxConnections];
      for (auto it = &m_sockets[0]; it != end; ++it) {
        if (!(*it)) {
          if (_handleRequest(client)) {
            *it = new WebSocket(client);
            ws = *it;
            if (_onConnection) _onConnection(*ws);
          }
          break;
        }
      }

      // Server is full ...
      if (!ws) _rejectRequest(client, WebSocketError::SERVICE_UNAVAILABLE);
    } else {
      ws->_readFrame(); // no need to check (ws) again, in line 74 was already checked to be non-null
    }
```
Please let me know if this change makes sense.

skaarj1989, 2021-02-10:
Hi @stevstrong,
I could not reproduce your issue on my **NUCLEO-L152RE** (at least with this [chat-client.htm](https://github.com/skaarj1989/mWebSockets/tree/dev/node.js))
1. Which **STM32** do you have?
2. Can you share your client code?
3. Please attach server debug log

stevstrong, 2021-02-10:
Hi David,

thanks for your reply.

The problem can be observed only if you do not send any message from the client to server in the first 5 seconds after establishing the socket channel.

I was using this example: https://github.com/stevstrong/mWebSockets/tree/master/examples/server-LEDcontrol, and in particular [this HTML file](https://github.com/stevstrong/mWebSockets/blob/master/examples/server-LEDcontrol/index.htm). It has no external dependencies, it is a very simple client which is mostly waiting for messages from the server and updates the status of the LED dependent on the received messages. Data will be only sent if you change the LED off or on time.

stevstrong, 2021-02-10:
Btw, the same error case could be reproduced with your client:
[image]
(the second line comes from my server)

skaarj1989, 2021-02-10:
In my case, disconnection happens for every 2nd client. You are right, that little `else` makes difference ;)
I'll update master branch if autobahn testsuite will show me green for every supported platform.
Thanks for your contribution.

stevstrong, 2021-02-10:
Welcome, I think this solves eventually some closed but not 100% understood issue, I hope that other can benefit from this.

skaarj1989, 2021-02-25:
Master branch updated.

closed 2021-02-25 by skaarj1989