skaarj1989/mWebSockets #33: websocket.listen() blocking loop()

andmim, 2021-03-07:
Good evening,

I am trying to use your library to connect my ESP32 and a W5500 to my server which is running a node.js script to receive sensor data. The connection is established successfully, however the `listen()` function is taking up a lot of time in my main.

```
void Setup() {

....

Ethernet.begin(macaddress);

webSocket.onOpen(WSonOpen);
webSocket.onClose(WSonClose);
webSocket.onMessage(WSonMessage);
webSocket.open(ws_server, ws_port);

....

}

void loop() {
 ...

  Serial.printf("Before listen - %lu\n", millis());
  webSocket.listen();
  //printTime();
  Serial.printf("After listen - %lu\n", millis());

....

}
```

From my Serial output, my loop is being blocked for 5 Seconds.

```
Before listen - 391795
After listen - 396795
Reading...
Before listen - 396847
After listen - 401847
Reading...
Before listen - 401899
After listen - 406692
```

Is there anything I am missing, or do you have any other idea what may be causing it to take so long? I only found #10, but unfortunately, there was no resolution to that problem.

Best regards.

skaarj1989, 2021-03-22:
Hi @andmim,
Sorry for delayed reply, I have been quite busy working on another project.

To workaround this problem go into [WebSocketClient.cpp](https://github.com/skaarj1989/mWebSockets/blob/be8c11aad2e15d54e98aab5315cf67a954554e83/src/WebSocketClient.cpp#L52) and replace `_readFrame()` with:
```cpp
if (m_client.available()) _readFrame();
```
This is not the final solution though, I have some failed tests.
Once I figure it out I'll let you know and update master branch.

skaarj1989, 2021-06-16:
@andmim The problem has been resolved with the latest commit to master branch (all tests green).
(I will close this issue in the next month).

skaarj1989, 2021-07-01:
Looks like a dead issue.

closed 2021-07-01 by skaarj1989