skaarj1989/mWebSockets #37: Example simple-server.ino does not compile

fkromer, 2021-06-15:
I've installed `mWebScokets` via the Arduino IDE v2 Library Manager and tried to compile the example `examples/simple-server/simple-server.ino` for Arduino Mega2560. However there is thrown `Compilation error: Error: 2 UNKNOWN: exit status 1`.

skaarj1989, 2021-06-15:
Hi @fkromer 

Everything works fine for me.
[image]

1. Which Arduino IDE 2 version? **beta7**?
2. Do you have any older Arduino IDE installed? (1.8.*)
3. Does this error occur in any other example (like `Blink`)?
4. Have you selected a board from the list?
[image]

Also, try to enable verbose output for compilation in `File -> Preferences`

fkromer, 2021-06-15:
Hi @skaarj1989, it's

1.
[image]
2. Yes, I've tried 1.8.* as well. But it does not compile.
3. No
4. Yes
[image]

fkromer, 2021-06-15:
I've tried with Arduino IDE 1.8.* again. Compilation was successful. There was an error output related to Ethernet.h but I was able to upload. I'll try with Arduino IDE 2 again as well.

fkromer, 2021-06-15:
Arduino IDE 2 output is:

```
Using board 'mega' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for WebSocketServer.h: [mWebSockets@1.3.3]
ResolveLibrary(WebSocketServer.h)
  -> candidates: [mWebSockets@1.3.3]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for Ethernet.h: [Ethernet@2.0.0]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.0]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-IC:\\Program Files (x86)\\Arduino1-8-15\\libraries\\Ethernet\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-IC:\\Program Files (x86)\\Arduino1-8-15\\libraries\\Ethernet\\src" "c:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\WebSocket.cpp" -o nul
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-IC:\\Program Files (x86)\\Arduino1-8-15\\libraries\\Ethernet\\src" "c:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\WebSocketClient.cpp" -o nul
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-IC:\\Program Files (x86)\\Arduino1-8-15\\libraries\\Ethernet\\src" "c:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\WebSocketServer.cpp" -o nul
[... cut ...] \Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\utility.cpp" -o nul
Alternatives for Base64.h: []
ResolveLibrary(Base64.h)
  -> candidates: []
Using library mWebSockets at version 1.3.3 in folder: c:\Users\Florian\gitlab\hurrycane-arduino\libraries\mWebSockets 
Using library Ethernet at version 2.0.0 in folder: C:\Program Files (x86)\Arduino1-8-15\libraries\Ethernet 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

I've to uninstall Arduino 1.8.5 again probably...

fkromer, 2021-06-15:
Arduino IDE 2 + mWebSockets 1.3.3

```
Using board 'mega' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for WebSocketServer.h: [mWebSockets@1.3.3]
ResolveLibrary(WebSocketServer.h)
  -> candidates: [mWebSockets@1.3.3]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for Ethernet.h: []
ResolveLibrary(Ethernet.h)
  -> candidates: []
Using library mWebSockets at version 1.3.3 in folder: c:\Users\Florian\gitlab\hurrycane-arduino\libraries\mWebSockets 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

fkromer, 2021-06-15:
Arduino IDE 2 + mWebScokets 1.4.0

```
Using board 'mega' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for WebSocketServer.h: [mWebSockets@1.4.0]
ResolveLibrary(WebSocketServer.h)
  -> candidates: [mWebSockets@1.4.0]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-14857016F4B4D502F0F7EA668E09D3F4\\sketch\\wss.ino.cpp" -o nul
Alternatives for Ethernet.h: []
ResolveLibrary(Ethernet.h)
  -> candidates: []
Using library mWebSockets at version 1.4.0 in folder: c:\Users\Florian\gitlab\hurrycane-arduino\libraries\mWebSockets 
Compilation error: Error: 2 UNKNOWN: exit status 
```

fkromer, 2021-06-15:
@skaarj1989 As Arduino IDE v2 is beta I'm totally fine with using Arduino IDE 1.8.5 😏 Thanks for the help so far.

skaarj1989, 2021-06-15:
Looks like something is wrong with `Ethernet.h`
You have:
```
Alternatives for Ethernet.h: []
ResolveLibrary(Ethernet.h)
  -> candidates: []
```

And in my log:
```
Alternatives for Ethernet.h: [Ethernet@2.0.0]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.0]
```

fkromer, 2021-06-15:
Right. I already downgraded the Ethernet library and updated the library again to v2.0.0. However even with Arduini IDE v1.8.5 I've still an issue during compilation. It does compile but it's semms to work not perfectly well.

```
In file included from C:\Program Files (x86)\Arduino-1-8-15\libraries\Ethernet\src\Dns.cpp:8:0:
C:\Program Files (x86)\Arduino-1-8-15\libraries\Ethernet\src\Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':
C:\Program Files (x86)\Arduino-1-8-15\libraries\Ethernet\src\utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]
 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )
                      ~~~^~~
C:\Program Files (x86)\Arduino-1-8-15\libraries\Ethernet\src\Dns.cpp:164:18: note: in expansion of macro 'htons'
  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);
                  ^~~~~
Sketch uses 23366 bytes (9%) of program storage space. Maximum is 253952 bytes.
Global variables use 629 bytes (7%) of dynamic memory, leaving 7563 bytes for local variables. Maximum is 8192 bytes.
```

skaarj1989, 2021-06-15:
You don't have to worry about those warnings.
Does `simple-client` work after uploading to a board?

fkromer, 2021-06-16:
I cannot try yet cause I've no hardware for debugging.

fkromer, 2021-06-17:
I've tried with HW (Arduino Uno + Ethernet Shield v2). Unfortunatelly... it does not work. However I think this relates to the Arduino IDe v2 and not mWebSockets.

closed 2021-08-10 by stale[bot]