skaarj1989/mWebSockets #38: Feature request: subprotocol negotiation API

fkromer, 2021-06-15:
I've not found something related to [subprotocol negotiation](https://hpbn.co/websocket/#subprotocol-negotiation) in the sources. Is mWebSockets supporting it already? In case not you could of course expect either contribution or sponsoring (if this is an option for you).

skaarj1989, 2021-06-15:
It's just a proposal draft of what I might do.

Client: add additional parameter in `open()`, an array of subprotocols:
```cpp
WebSocketClient client;
const char *supportedProtocols[] = {"soap", "wamp", nullptr};
client.open("...", 3000, "/", supportedProtocols);
// The client will send:
// Sec-WebSocket-Protocol: soap, wamp
```

Server: add callback
```cpp
WebSocketServer server(3000);
server.setProtocolHandler([](const char **protocols) {
  // In this callback you are supposed to negotiate the subprotocol and return its name
  // (or nullptr on fail)

  // Iterate the list of protocols sent by the client
  for (int i = 0;; ++i) {
    const auto &protocol = protocols[i];
    if (protocol == nullptr) break;
  
    // The server will send: Sec-WebSocket-Protocol: soap
    if (strcmp(protocol, "soap") == 0) return "soap";
  }
  return nullptr; // supported protocol not found, terminates connection
});
```

fkromer, 2021-06-16:
I've not looked into implementations of other libs yet. Nevertheless... your idea sounds reasonable.

fkromer, 2021-06-16:
Without investing too much time I was able to find the relevant sources in Pythons `websockets`. The utility function verifying the subprotocols is [/src/websockets/client.py#L243-L274](https://github.com/aaugustin/websockets/blob/d6188e71df9b5cf6dabadb352ac0ce489e4408cf/src/websockets/client.py#L243-L274). It checks the subprotocols in the response header. The verification is called in the client code in [src/websockets/client.py#L169](https://github.com/aaugustin/websockets/blob/d6188e71df9b5cf6dabadb352ac0ce489e4408cf/src/websockets/client.py#L169) as well as in the server code in [src/websockets/server.py#L229](https://github.com/aaugustin/websockets/blob/d6188e71df9b5cf6dabadb352ac0ce489e4408cf/src/websockets/server.py#L229). In both cases (client as well as server) the function is triggered after a handshake request has been received.

fkromer, 2021-06-16:
In case of Pythons `websockets` the subprotocols may be defined API-wise via the [client connection class instance constructor](https://github.com/aaugustin/websockets/blob/d6188e71df9b5cf6dabadb352ac0ce489e4408cf/src/websockets/client.py#L47-L53) as well as the [server class instance constructur](https://github.com/aaugustin/websockets/blob/d6188e71df9b5cf6dabadb352ac0ce489e4408cf/src/websockets/server.py#L51-L59).

skaarj1989, 2021-06-16:
Initial implementation for subprotocol negotiation has been uploaded to master branch.
Instead of array of values I decided to use comma separated values.

Client side:
```cpp
WebSocketClient client;
client.open("192.168.46.31", 3000, "/", "foo,bar,baz");
const auto protocol = client.getProtocol();
if (protocol) {
  // do something about it ...
}
```
> Additionally, you have an option to query protocol straight out of a WebSocket via `getProtocol()` (might return nullptr)

Server side:
```cpp
WebSocketServer server{3000};
wss.begin(nullptr, [](const char *protocols) {
  // Iterate csv with strtok or whatever you like ...
  // ...

  return "foo";
});

wss.onConnection([](WebSocket &ws) {
  const auto protocol = ws.getProtocol();
  if (protocol) {
    // whatever ...
  }
  // connect callbacks ...
});
```

fkromer, 2021-06-17:
Wow, great. Thanks a lot.

closed 2021-07-24 by stale[bot]