skaarj1989/mWebSockets #43: 'class net::WebSocketServer' has no member named 'getProtocol' - simple-server.ino not compiling

Perfectmistake, 2021-06-29:
This seems to be a similar issue to what has been discussed previously, but currently I am trying to compile simple-server.ino via Arduino IDE v1.8.13. However when I compile, this line gets flagged:
```
const auto protocol = ws.getProtocol();
```
The full verbose output is shown below:
```
Arduino: 1.8.13 (Linux), Board: "Arduino Uno"

/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware -tools /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/tools-builder -tools /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -built-in-libraries /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries -libraries /home/tsmart/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path /tmp/arduino_build_49149 -warnings=none -build-cache /tmp/arduino_cache_198035 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -verbose /home/tsmart/Desktop/code/Arduino/mWebSocketServer/mWebSocketServer.ino
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware -tools /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/tools-builder -tools /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -built-in-libraries /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries -libraries /home/tsmart/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path /tmp/arduino_build_49149 -warnings=none -build-cache /tmp/arduino_cache_198035 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr -verbose /home/tsmart/Desktop/code/Arduino/mWebSocketServer/mWebSocketServer.ino
Using board 'uno' from platform in folder: /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr
Detecting libraries used...
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard /tmp/arduino_build_49149/sketch/mWebSocketServer.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for WebSocketServer.h: [mWebSockets@1.4.0]
ResolveLibrary(WebSocketServer.h)
  -> candidates: [mWebSockets@1.4.0]
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src /tmp/arduino_build_49149/sketch/mWebSocketServer.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Ethernet.h: [UIPEthernet@2.0.10 Ethernet@2.0.0]
ResolveLibrary(Ethernet.h)
  -> candidates: [UIPEthernet@2.0.10 Ethernet@2.0.0]
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet/src /tmp/arduino_build_49149/sketch/mWebSocketServer.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet/src /home/tsmart/Arduino/libraries/mWebSockets/src/CryptoLegacy/BlockCipher.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet/src /home/tsmart/Arduino/libraries/mWebSockets/src/CryptoLegacy/Cipher.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet/src /home/tsmart/Arduino/libraries/mWebSockets/src/CryptoLegacy/Crypto.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/cores/arduino -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/variants/standard -I/home/tsmart/Arduino/libraries/mWebSockets/src -I/home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet/src /home/tsmart/Arduino/libraries/mWebSockets/src/CryptoLegacy/Hash.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
[... cut ...]
Using library mWebSockets at version 1.4.0 in folder: /home/tsmart/Arduino/libraries/mWebSockets 
Using library Ethernet at version 2.0.0 in folder: /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/libraries/Ethernet 
Using library SPI at version 1.0 in folder: /home/tsmart/Desktop/code/Arduino/arduino-1.8.13/hardware/arduino/avr/libraries/SPI 
exit status 1
'class net::WebSocket' has no member named 'getProtocol'

```
I wish to upload this sketch to an Arduino Uno (ATmega328) with a W5500 Ethernet Shield. How do I fix this issue?
Thanks in advance.

skaarj1989, 2021-06-29:
@Perfectmistake It's a preview feature, use master branch instead of **1.4.0**

Perfectmistake, 2021-06-29:
Thank you very much!

closed 2021-06-29 by Perfectmistake