skaarj1989/mWebSockets #45: bug in mega2560

C4psL0cks, 2021-07-02:
When using the original, there will be problems. in mega2560

//  Serial.println(F("Initializing ... "));
//
//  Ethernet.init(53); // Mega2560
//  Ethernet.begin(mac);
//
//  Serial.print(F("Device IP: "));
//  Serial.println(Ethernet.localIP());

but. use 

#include "SPI.h"
#include "Ethernet.h"

#include <WebSocketClient.h>
using namespace net;

#define _SERIAL Serial

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

WebSocketClient client;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.begin(115200);
  Serial.println("Starting...");

  // start the Ethernet connection:
  while (Ethernet.begin(mac) != 1){
    Serial.print(".");
  }
  Serial.print("My IP :");
  Serial.println(Ethernet.localIP());
  
  client.onOpen([](WebSocket & ws) {
    Serial.println(F("Type a message in the following format: <text>"));
    Serial.println(F("----------------------------------------------"));
  });

  client.onMessage(
    [](WebSocket & ws, const WebSocket::DataType, const char *message,
  uint16_t) {
    Serial.println(message);
  });
  client.onClose([](WebSocket &, const WebSocket::CloseCode, const char *,
  uint16_t) {
    Serial.println(F("Disconnected"));
  });

  if (!client.open("192.168.0.104", 3000)) {
    Serial.println(F("Connection failed!"));
    while (true)
      ;
  }
}

void loop() {

  client.listen();

}

***** 

21:39:09.616 -> Starting...
21:39:15.312 -> My IP :192.168.0.102
21:39:15.357 -> Type a message in the following format: <text>
21:39:15.357 -> ----------------------------------------------
21:39:15.357 -> Hello from Node.js

I think there should be a problem in the library where you defined the #include's contact "SPI.h".
#include "Ethernet.h"

skaarj1989, 2021-07-02:
Sorry, but I don't understand what you are talking about. What kind of problem do you have?

C4psL0cks, 2021-07-02:
When I use your example It can't contact mega2560 at all, but when I rewrite the code it can contact mega2560. I think your code might have a problem writing your src in contact with libarry.

C4psL0cks, 2021-07-02:
The problematic part is here.

  _SERIAL.println(F("Initializing ... "));

  // Ethernet.init(10);
  Ethernet.init(53); // Mega2560
  // Ethernet.init(5); // ESPDUINO-32
  // Ethernet.init(PA4); // STM32

  Ethernet.begin(mac); //, ip);

  _SERIAL.print(F("Device IP: "));
  _SERIAL.println(Ethernet.localIP());
  
  ----------------------------------------------
22:22:50.276 -> Initializin.0.0.0:3000
22:22:50.276 -> Initializing ... 
22:22:50.827 -> Server running at 0.0.0.0:3000

  
  but when i change your code in example to this format
  
#include "SPI.h"
#include "Ethernet.h"
#include <WebSocketClient.h>
using namespace net;
byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

WebSocketClient client;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.begin(115200);
  Serial.println("Starting...");

  // start the Ethernet connection:
  while (Ethernet.begin(mac) != 1){
    Serial.print(".");
  }
  Serial.print("My IP :");
  Serial.println(Ethernet.localIP());
  }

-------------------------------------

21:39:09.616 -> Starting...
21:39:15.312 -> My IP :192.168.0.102
21:39:15.357 -> Type a message in the following format: <text>
21:39:15.357 -> ----------------------------------------------
21:39:15.357 -> Hello from Node.js

C4psL0cks, 2021-07-02:
I think it's a problem I'm here to inform you, but if you think it's not a problem and you don't care about it, I'm happy to apologize. With all due respect, I see that your code is very useful for others and for me. When there is a problem, I come to inform you. Thank you.

skaarj1989, 2021-07-02:
It's ok.
The only problem is that, in some circumstances, you have to comment: `Ethernet.init(...)`

closed 2021-07-04 by skaarj1989