skaarj1989/mWebSockets #47: Unreliable connection setup

fkromer, 2021-08-09:
**Describe the bug**

If I try to connect from a client to a websocket server implemented with mWebSockets the connection is not always setup.

**Environment info**
- IDE w/version: Arduino IDE 1.8.15
- Platform/Board: Arduino Due
- Network controller (shield or module): EthernetShield 2

**Expected behavior**

The connection setup is executed properly every time I try to connect.

**Failure log**

No failure log output in case of connection is not setup (`#define _DUMP_HANDSHAKE` is set).

Log in case connection is setup:

```
[Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.1.2
[Line #2] User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:90.0) Gecko/20100101 Firefox/90.0
[Line #3] Accept: */*
[Line #4] Accept-Language: de,en-US;q=0.7,en;q=0.3
[Line #5] Accept-Encoding: gzip, deflate
[Line #6] Sec-WebSocket-Version: 13
[Line #7] Origin: moz-extension://d4825527-04cb-4cc5-9a5a-cca3f570223a
[Line #8] Sec-WebSocket-Extensions: permessage-deflate
[Line #9] Sec-WebSocket-Key: On6QMBBNL3JfSru3nICLsg==
[Line #10] Connection: keep-alive, Upgrade
[Line #11] Pragma: no-cache
[Line #12] Cache-Control: no-cache
[Line #13] Upgrade: websocket
[Line #14]
```

**Additional context**

Firecamp log output in case connection is not setup:

[image]

Firecamp log output in case connection is setup:

[image]

skaarj1989, 2021-08-10:
Hi @fkromer
Thanks for using the issue template ;)
How often does this problem occur? And how many seconds passes between `connecting` and `error`?

Anyways, this means that your client didn't even make it to `_handleRequest` which suggests that Arduino did'nt notice a client at all.

fkromer, 2021-08-10:
Hey @skaarj1989, you are welcome 😉 The error is displayed "immediatelly" (within a fraction of a second). Right, that's the strange thing...

fkromer, 2021-10-05:
@skaarj1989 The behavior is reproducible with [Postman beta websocket support](https://blog.postman.com/postman-supports-websocket-apis/).

skaarj1989, 2021-10-05:
@fkromer Have you tried to add some delay before `open()`? like 1-2 sec?
Last month I replaced my router which introduced some issues with the connection and that 1sec delay resolved it.

About your Ethernet shield, is it "official" or a clone?

fkromer, 2021-10-05:
I'm waiting more than 5 seconds after power on before I try to connect to my server with a client. ATM I'm not using a router/switch but connect directly via crossover cable. The behaviour when connecting via a router is not different. It's an official Ethernet shiled v2.

fkromer, 2021-10-05:
@skaarj1989 It seems like one cannot use port 80 with websockets and a restful api server at the same time 😄  Can I workaround this somehow?

skaarj1989, 2021-10-05:
Is it really a big deal for you to a share REST server with websocket one?
If I were you, I would use two different ports like: **80** and **8080**

fkromer, 2021-10-05:
@skaarj1989 Sometimes port 8080 is used by a proxy server or caching server. I'd like to avoid using it. However in case there is no workaround I'll use it.

closed 2021-10-07 by fkromer