skaarj1989/mWebSockets #51: note: array types have different bounds

guilhermeaiolfi, 2021-09-29:
**Describe the bug**
Hi, I am trying to use this library but I am getting the following message after compiling:
```
WebSocketServer.h:14:7: note: array types have different bounds
 class WebSocketServer {
       ^
```

**Environment info**
- IDE 1.8.16
- Arduino Uno
- WiFi
- mWebSockets 1.4.0

Should I be worried about it?

skaarj1989, 2021-09-30:
@guilhermeaiolfi It's fine. You can also use **master** version (I think it's time to create a new release).

guilhermeaiolfi, 2021-09-30:
That would be nice. I'm just started playing with arduíno (yesterday was the first day) and I am using the library manager. I guess I could manually use the master code but I have a lot to figure it out yet. BTW, thank you for this library, it is exactly what I wanted it.

closed 2021-11-06 by stale[bot]