skaarj1989/mWebSockets #55: NETWORK_CONTROLLER switched by user

Sampozzo, 2021-11-10:
Hello, thank you very much for this library, i'm using it in my project and it works very well.

I'm building an ESP32 device with an angular client that talks with the controller on websocket and it works well over ethernet, over wifi and over softAP.
My device at boot load the configuration from SPIFFS and start. 

What i'm trying to do now is to add the ability for the user to choose the connection type (Eth/softAP/Wifi), save it in config and reboot with the new configuration. 

But i think it is not possible because NETWORK_CONTROLLER must be defined in the config.h and can't be changed. 
Any idea how i can do?

Thank you very much and sorry for my english.

skaarj1989, 2021-11-11:
Hi @Sampozzo

> (...) NETWORK_CONTROLLER must be defined in the config.h and can't be changed later. Any idea how i can do?

This is correct, you have to specify network controller at compile time. This can't be changed without major refactoring.

Sampozzo, 2021-11-11:
Hi, thank you for your reply!

i'm using platform.io and i did those steps:

1. moved the library from the libdeps folder to lib folder and removed from platformio.ini
2. made a copy of the library, named mWebSocketWifi
3. changed the second namespace to netWifi
4. solved some little compile errors
5. declared EthernetClient ethClient; and WiFiClient wifiClient; as globals
6. in setup, read config from SPIFFS and start ethernet client OR wificlient 

now i can switch from wifi to ethernet without reflash. 
I know this isn't the way, but it's works and i'm only learning :) next step is to go deeper in your library to learn how to write single one :)

Thank you very much!

closed 2021-12-18 by stale[bot]