skaarj1989/mWebSockets #56: can't send from server Payload > 2048

Sampozzo, 2021-11-18:
**Describe the bug**

On my MwebSockets server i can't send payload > 2048
On varios javascript client i have the error 

```
WebSocketClient.html:37 WebSocket connection to 'ws://192.168.4.30:3000/' failed: Could not decode a text frame as UTF-8.
```

**Environment info**
- IDE w/version [platform.io vs code]
- Platform/Board: [ESP32]
- Network controller (shield or module): [W5500 or wifi]

**Expected behavior**
If my payload is <= 2048 the client receive the message

**Failure log**
```
[Line #0] GET / HTTP/1.1
[Line #1] Host: 192.168.4.30:3000
[Line #2] Connection: Upgrade
[Line #3] Pragma: no-cache
[Line #4] Cache-Control: no-cache
[Line #5] User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/96.0.4664.45 Safari/537.36
[Line #6] Upgrade: websocket
[Line #7] Origin: null
[Line #8] Sec-WebSocket-Version: 13
[Line #9] Accept-Encoding: gzip, deflate
[Line #10] Accept-Language: it,en-GB;q=0.9,en;q=0.8,it-IT;q=0.7
[Line #11] Sec-WebSocket-Key: bEXwqjPKVFCpFdrTY2doLA==
[Line #12] Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
[Line #13]
New client: 192.168.5.25
TX FRAME : OPCODE=1, FIN=True, RSV=0, PAYLOAD-LEN=2049, MASK=None
{"name":"xxxxxxxx","network":"ethernet","wifi_sid":"iot","wifi_password":"wifipwd","softap_password":"12345678","dhcp":0,"ip":"192.168.4.30","subnet":"255.255.255.0","gateway":"192.168.5.1","dns":"192.168.5.4","mqtt":0,"mqtt_ip":"192.168.1.35","mqtt_user"TX BYTES = 2053
```

**Additional context**

i have on the config.h

```
constexpr uint16_t kBufferMaxSize{4096};
```

Sampozzo, 2021-11-18:
Update: over wifi network it seems to works.
The problem is over ethernet (W5500).

Any idea what can I do?

Thank you

Sampozzo, 2021-11-18:
OK, the problem is on the ethernet.h library that is limited to 2048bytes. 
You need to reduce MAX_SOCK_NUM and uncomment #define ETHERNET_LARGE_BUFFERS.

Now it is working.

Thank you again for your great work.

skaarj1989, 2021-11-18:
Hi @Sampozzo I'm glad that you resolved your own issue ;) I think it's something worth mentioning in the `README` as it might bite someone in the future too.

closed 2021-12-25 by stale[bot]