skaarj1989/mWebSockets #57: 'SerialUSB' was not declared in this scope

LaudixGit, 2021-12-29:
**Describe the bug**
Compile fails

Using the unmodified "simple-server" example

**Environment info**
- IDE w/version: Arduino 1.8.15
- Platform/Board: Adafruit Feather M4 Express (SAMD51)
- Network controller (shield or module): 
- mWebSockets Version: 1.5.1

**Expected behavior**
SAMD51 should be detected correctly

**Failure log**
```
/home/laudix/Arduino/libraries/mWebSockets/src/utility.cpp: In function 'void printf(const __FlashStringHelper*, ...)':
/home/laudix/Arduino/libraries/mWebSockets/src/utility.cpp:20:3: error: 'SerialUSB' was not declared in this scope
   SerialUSB.print(buffer);
   ^~~~~~~~~
/home/laudix/Arduino/libraries/mWebSockets/src/utility.cpp:20:3: note: suggested alternative: 'Serial'
   SerialUSB.print(buffer);
   ^~~~~~~~~
   Serial
Multiple libraries were found for "Ethernet.h"
 Used: /opt/arduino-1.8.15/libraries/Ethernet
 Not used: /home/laudix/Arduino/libraries/UIPEthernet
Multiple libraries were found for "Adafruit_ZeroDMA.h"
 Used: /home/laudix/.arduino15/packages/adafruit/hardware/samd/1.7.5/libraries/Adafruit_ZeroDMA
 Not used: /home/laudix/Arduino/libraries/Adafruit_Zero_DMA_Library
exit status 1
Error compiling for board Adafruit Feather M4 Express (SAMD51).

```

**Additional context**
editing platform.h (as [suggested](https://github.com/skaarj1989/mWebSockets/issues/19)) didn't repair so reverted back

LaudixGit, 2021-12-29:
note: forcing the use of Serial doesn't help because utility.cpp neglects to use the _SERIAL value

`
#if PLATFORM_ARCH == PLATFORM_ARCHITECTURE_SAMD21**x**
#  define _SERIAL SerialUSB
#else
#  define _SERIAL Serial
#endif
`

LaudixGit, 2021-12-29:
Compiles when I comment-out line 20 of
/Arduino/libraries/mWebSockets/src/utility.cpp

skaarj1989, 2021-12-31:
Can't find `Adafruit Feather M4 Express (SAMD51)` in Arduino Board Manager, where did you get it?

LaudixGit, 2021-12-31:
@skaarj1989  The Board Installer is part of a Adafruit package.
There's [the Adafruit sales page](https://www.adafruit.com/product/3857), and 
I followed [their instructions](https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51/setup)
to add [their board ](https://adafruit.github.io/arduino-board-index/package_adafruit_index.json)to the IDE via the "Additional Boards Manager" in the preferences settings.

The enabled me to use Board Manager to install both "Arduino SAMD Boards (32-Bits ARM Cortex-M0+"  and "Adafruit SAMD Boards", and finally, 
select Adafruit SAMD (32-bits ARM Cortex-M0+ and Cortex-M4) Boards **/**  Adafruit Feather M4 Express (SAMD51)

A little more research this morning to [list the variables](https://arduino.stackexchange.com/questions/19892/list-of-arduino-board-preprocessor-defines) with
        `grep board= `find . -name boards.txt` | cut -f2 -d= | sort -u`
And I think the relevant IDE variable, to identify the board, is "FEATHER_M4"
(but that is a guess, I am not an expert)

skaarj1989, 2021-12-31:
I will submit a fix for this soon.

For a quick workaround, you can change `SerialUSB` to `Serial`. This will give you some debug information (in opposition to comment-out)
Anyway, have you managed to get it work?

LaudixGit, 2021-12-31:
Yes. I have completed my task.  But it will be nice to have the your update so future upgrades will "just work" - no modifications needed