skaarj1989/mWebSockets #58: The used network controller should be configurable from the sketch

Anton-V-K, 2022-02-19:
The approach with selecting the network controller inside the library's `config.h` is hardly usable in real projects - it is impossible to tweak it for every build.

There are several approaches which can address this issue:
1. Wrap `#define NETWORK_CONTROLLER ETHERNET_CONTROLLER_W5X00` with `#ifnded NETWORK_CONTROLLER` and `#endif` so if the build system support global defines, the project will choose the proper network controller.
2. Do a meaningful guess of the network controller (e.g., when `ESP8266` or `ESP32` is defined, the chances are high that WiFi is used), so for rigid build systems (like the one in Arduino IDE) the library can be used as is.

**Environment info**
- IDE w/version: Arduino IDE 1.8.16
- Platform/Board: esp8266
- Network controller (shield or module): WiFi

skaarj1989, 2022-02-20:
I've been thinking about refactoring into templates.
No more `#define NETWORK_CONTROLLER` in `config.h`. Instead, you will need to specialize the client/server.

```cpp
WebSocketServer<WifiServer, WiFiClient> server;
WebSocketClient<EthernetClient> client;
```

Additionally you could use multiple network controllers in a single (more powerful) board.

skaarj1989, 2022-02-20:
Initial commit pushed into the [meta](https://github.com/skaarj1989/mWebSockets/tree/meta) branch.

skaarj1989, 2023-12-04:
@infrafast

1. Yes
2. No
3. Yes

skaarj1989, 2023-12-04:
Why do you insist on merging to `master`? Is there anything that prevents you from using the `meta` branch?
Using multiple network controllers on a single MCU is an edge case. In two years you are the second one who asked for that feature.

BTW the library (also the `meta` branch) is tested with [autobahn testsuite](https://github.com/crossbario/autobahn-testsuite).

skaarj1989, 2023-12-04:
Looks like a branch/tag should be put after `#` character, so you could try:
`https://github.com/skaarj1989/mWebSockets#meta`

infrafast, 2023-12-04:
great, I am able to compile using the meta branch.
things works well, when I use WiFi but if I force the usage of ethernet using the if directive, I got bellow error message when compiling, would you have any idea why?

.pio/libdeps/proto/mWebSockets/src/WebSocket.hpp:298:28:  error: passing 'const EthernetClient' as 'this' argument discards qualifiers [-fpermissive]

#if 0
using NetClient = WiFiClient;
using NetServer = WiFiServer;
#else
using NetClient = EthernetClient;
using NetServer = EthernetServer;
#endif

skaarj1989, 2023-12-04:
Which MCU?

infrafast, 2023-12-05:
ESP32

skaarj1989, 2023-12-05:
Fixed.
Could you confirm?

infrafast, 2023-12-05:
Yes, it works, thanks a lot for your work and support. I understood that my case to have both WiFi and Eth is not the most common so I don't want to bother. Still, would you route me to a possible solution to instantiate the proper object (Eth / WiFi) at run time and not at compilation time,  as simple example ?
I was thinking about creating a pointer to WebSocketServer  and then be able to use the same code however I don't know how to do that with/without template ?

Anton-V-K, 2023-12-05:
> possible solution to instantiate the proper object (Eth / WiFi) at run time and not at compilation time, as simple example ?

I believe, this is a good question for a separate [discussion](https://github.com/skaarj1989/mWebSockets/discussions) :)

infrafast, 2023-12-08:
OK, added the discussion, I think this issue can therefore be closed ?