skaarj1989/mWebSockets #61: use build_flags on platformio instead of editing config.h

paulkilroy, 2023-02-15:
If you're using platformio and have this library included in dep_libs it will automatically download and compile with it (very nice and easy). Then, I can edit the config.h in the dep_libs directory...

But I think it would be easier to just edit the platformio.ini file with a -DNETWORK_CONTROLLER=3, then all your platform specific edits are in one place. But to do this, it would require you to change config.h to have an #ifndef NETWORK_CONTROLLER around it so the compiler doesn't get upset about a redefined #define. The upshot of this is that its completely backwards compatible with your current scheme.

Thanks

skaarj1989, 2023-02-15:
At the moment I don't have **PlatformIO**, when I fix this, can you use the **master** branch, or do you need a new release?

paulkilroy, 2023-02-17:
Sure - Happy to help, but I apologize but I’m still a bit new to Git. I can
include any library based on the following structure/syntax in PlatformIO.
Hope that helps.

pio lib [STORAGE_OPTIONS] install (with no args, install project
dependencies from "lib_deps")
pio lib [STORAGE_OPTIONS] install <ownername/name>
pio lib [STORAGE_OPTIONS] install <ownername/name>@<version>
pio lib [STORAGE_OPTIONS] install <ownername/name>@<version range>
pio lib [STORAGE_OPTIONS] install <zip or tarball url>
pio lib [STORAGE_OPTIONS] install file://<zip or tarball file>
pio lib [STORAGE_OPTIONS] install file://<folder>
pio lib [STORAGE_OPTIONS] install <repository>
pio lib [STORAGE_OPTIONS] install <name>=<repository> (name it should
have locally)
pio lib [STORAGE_OPTIONS] install <repository#tag> ("tag" can be
commit, branch or tag)

By the way, the reason I switched to PlatformIO is the WokWi simulator.
It’s pretty amazing and can run on the web, but also inside
VSCode/PlatformIO. I can run / test / debug all in that environment and
can’t go back to Arduino IDE.

skaarj1989, 2023-02-17:
Done.

paulkilroy, 2023-02-18:
Works great - Thanks!

closed 2023-02-18 by skaarj1989