skaarj1989/mWebSockets #62: Solution

gigaj0ule, 2023-02-21:
I found a better solution

We must #include <strings.h> at the top of utility.h

It's not guaranteed to be automatically included in every arduino environment's "platform.h"

In some, it is. In other environments, it's not.

```
#pragma once

#include "platform.h"
#include <strings.h>

#define SAFE_DELETE(ptr)                                                       \
  {                                                                            \
    if (ptr != nullptr) {                                                      \
      delete ptr;                                                              \
      ptr = nullptr;                                                           \
    }                                                                          \
  }

#define SAFE_DELETE_ARRAY(ptr)                                                 \
  {                                                                            \
    if (ptr != nullptr) {                                                      \
      delete[] ptr;                                                            \
      ptr = nullptr;                                                           \
    }                                                                          \
  }

#ifdef _DEBUG
#  define __debugOutput printf
#else
#  define __debugOutput(...)
#endif

void printf(const __FlashStringHelper *fmt, ...);

namespace net {

IPAddress fetchRemoteIp(const NetClient &);

} // namespace net

```

gigaj0ule, 2023-02-21:
We can also put it in "platform.h" just under stdint.h

```
#pragma once

/** @file */

#include <stdint.h>
#include <strings.h>

//
// Board architectures:
//

/** @cond */
#define PLATFORM_ARCHITECTURE_AVR 1
#define PLATFORM_ARCHITECTURE_ESP8266 2
#define PLATFORM_ARCHITECTURE_ESP32 3
#define PLATFORM_ARCHITECTURE_SAM 4
#define PLATFORM_ARCHITECTURE_SAMD21 5
#define PLATFORM_ARCHITECTURE_STM32 6
/** @endcond */

#if defined(__AVR__)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_AVR
#elif defined(ESP8266)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_ESP8266
#elif defined(ESP32)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_ESP32
#elif defined(ARDUINO_ARCH_SAM)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_SAM
#elif defined(ARDUINO_ARCH_SAMD)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_SAMD21
#elif defined(ARDUINO_ARCH_STM32)
#  define PLATFORM_ARCH PLATFORM_ARCHITECTURE_STM32
#else
#  error "Unsupported platform"
#endif

//
// Ethernet/WiFi devices:
//

/** @cond */
#define ETHERNET_CONTROLLER_W5X00 1
#define ETHERNET_CONTROLLER_ENC28J60 2
#define NETWORK_CONTROLLER_WIFI 3
/** @endcond */

#include "config.h"

#if NETWORK_CONTROLLER == ETHERNET_CONTROLLER_W5X00
#  include <Ethernet.h>
/** Specifies maximum number of clients connected to server. */
constexpr uint8_t kMaxConnections{MAX_SOCK_NUM};
#elif NETWORK_CONTROLLER == ETHERNET_CONTROLLER_ENC28J60
#  include <EthernetENC.h>
constexpr uint8_t kMaxConnections{4};
#elif NETWORK_CONTROLLER == NETWORK_CONTROLLER_WIFI
#  if PLATFORM_ARCH == PLATFORM_ARCHITECTURE_ESP8266
#    include <ESP8266WiFi.h>
#  else
#    include <WiFi.h>
#  endif
#  include <WiFiClient.h>
#  include <WiFiServer.h>
constexpr uint8_t kMaxConnections{8};
#else
#  error "Network controller is required!"
#endif

/** @cond */
#if NETWORK_CONTROLLER == NETWORK_CONTROLLER_WIFI
using NetClient = WiFiClient;
using NetServer = WiFiServer;
#else
using NetClient = EthernetClient;
using NetServer = EthernetServer;
#endif
/** @endcond */

/**
 * @def PLATFORM_ARCH
 * @def NETWORK_CONTROLLER
 */
```

skaarj1989, 2023-02-21:
You mean **string.h**?

```
fatal error: strings.h: No such file or directory
 #include <strings.h>
```

gigaj0ule, 2023-02-21:
Huh, now this is interesting...

<string.h> does not contain strcasecmp, but <strings.h> does

https://stackoverflow.com/questions/4291149/difference-between-string-h-and-strings-h

<strings.h> is not included in your environment, and that is puzzling since arduino uses GCC under the hood, and <strings.h> is part of the GCC distribution!

Maybe there is a -D flag we can use to detect the compiler environment and optionally include <strings.h>

skaarj1989, 2023-02-21:
There is a solution (requires C++17)

```cpp
#if __has_include("strings.h")
# include <strings.h>
#endif
```

closed 2023-03-30 by stale[bot]