skaarj1989/mWebSockets #63: Cant Compile on ESP32 dev module

affix-development, 2023-03-03:
**bug**
Cant compile a client example with Esp32 and W5500 module SPI interface.

**Environment info**
- IDE w/version [e.g Arduino IDE 1.8.12]
- Platform/Board: ESP32 dev module
- Network controller (shield or module): W5500 Module SPI

**Expected behavior**
A clear and concise description of what you expected to happen.

**Failure log**
```
libraries\mWebSockets-master\src\WebSocketServer.h:104:13: error: cannot declare field 'net::WebSocketServer::m_server' to be of abstract type 'EthernetServer'
NetServer m_server;

In file included from ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\platform.h:49:0,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\utility.h:3,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocket.h:5,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocketServer.h:5,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocketServer.cpp:1:
\ArduinoIDE\libraries\Ethernet\src/Ethernet.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {

In file included from \ArduinoIDE\portable\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:152:0,
                 \ArduinoIDE\libraries\Ethernet\src/Ethernet.h:51,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\platform.h:49,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\utility.h:3,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocket.h:5,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocketServer.h:5,
                 \ArduinoIDE\portable\sketchbook\libraries\mWebSockets-master\src\WebSocketServer.cpp:1:
\ArduinoIDE\portable\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;

```

My Scketch

```

#include <soc/spi_pins.h>
#include <SPI.h>
#include <WebSocketClient.h>
using namespace net;
//EthernetClient client;

WebSocketClient client;

#define _SERIAL Serial

#define DEBUG_ETHERNET_GENERIC_PORT         Serial
#define USE_W5100                           false

#define ETHERNET_USE_ESP32
#define BOARD_TYPE      "ESP32"

#define W5500_RST_PORT   22
#define PIN_MISO          MISO
#define PIN_MOSI          MOSI
#define PIN_SCK           SCK
#define PIN_SS            SS
#define SHIELD_TYPE       "W5x00 using Ethernet_Generic Library on SPI"
#define USE_THIS_SS_PIN   5
// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
//byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
uint8_t mac[6]; // array for mac address

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }
  //WiFi.macAddress(mac);
  esp_efuse_mac_get_default(mac);

  Ethernet.init (USE_THIS_SS_PIN);
  Ethernet.begin(mac);
  Serial.println("=========================");
  Serial.println("Currently Used SPI pinout:");
  Serial.print("MOSI:");
  Serial.println(PIN_MOSI);
  Serial.print("MISO:");
  Serial.println(PIN_MISO);
  Serial.print("SCK:");
  Serial.println(PIN_SCK);
  Serial.print("SS:");
  Serial.println(USE_THIS_SS_PIN);
  Serial.println(F("========================="));

  Serial.print(F("Connected! IP address: "));
  Serial.println(Ethernet.localIP());

 
  if (!client.open("192.168.46.31", 3000, "/")) {
    _SERIAL.println(F("Connection failed!"));
    while (true)
      ;
  }
  
  
}

void loop()
{
    client.listen();
  delay(10);
}

```

skaarj1989, 2023-03-03:
Have you tried this: https://github.com/arduino-libraries/Ethernet/issues/88#issuecomment-455498941 ?

affix-development, 2023-03-03:
I delete the Ethernet Library and its compile well.

But now I Cant connect with socketServer(Using as client scketch).
And DHCP not works too.

I dont know what do now.

skaarj1989, 2023-03-03:
You need an Ethernet library (try to reinstall it).

closed 2023-04-11 by stale[bot]