skaarj1989/mWebSockets #64: TinyGSM

trickyau, 2023-08-21:
Any idea if it's possible to use TinyGSM library with this library?

skaarj1989, 2023-08-21:
I managed to compile with **TinyGSM**, but I cannot test it.

How to make it work?
- In `platform.h` add **NETWORK_CONTROLLER_GSM**, conditionally define GSM Modem and `#include <TinyGsmClient.h>` `using NetClient = TinyGsmClient;`
- Looks like **TinyGSM** is a client only library. Remove **WebSocketServer**.
- In `fetchRemoteIp` from `utility.cpp`, return `IPAddress()` if GSM is used. Unfortunately,**TinyGSM** uses a string instead of **IPAddress**.
- Add an overloaded constructor to **WebSocketClient**:
```cpp
#if NETWORK_CONTROLLER == NETWORK_CONTROLLER_GSM
  WebSocketClient(const NetClient &client) { m_client = client; }
#endif
```

Finally you can use it this way:
```cpp
TinyGsm modem(SerialAT);
WebSocketClient client(TinyGsmClient{modem});
```

trickyau, 2023-08-21:
Sorry if this is a stupid question, but where exactly should overloaded constructor be placed?

If I place it it before or after default lines I get an compile error expecting 0 arguments, 1 provided.

`  WebSocketClient() = default;
  ~WebSocketClient() = default;`

skaarj1989, 2023-08-21:
I would add it between the default constructor and the destructor.
https://github.com/skaarj1989/mWebSockets/blob/eeeb71f06048b4a0807113df5fc302d044be5368/src/WebSocketClient.h#L19

trickyau, 2023-08-21:
I'm probably doing something wrong on my end, but I place overload between L18 and L19 I still run into same issue.

WebSocketClient.h:19:3: note: candidate: 'net::WebSocketClient::WebSocketClient()'
WebSocketClient.h:19:3: note:   candidate expects 0 arguments, 1 provided

I'm also seeing this which I assume is probably related the above.

error: no matching function for call to 'net::WebSocketClient::WebSocketClient(TinyGsmClient)'
 WebSocketClient client(TinyGsmClient{modem});

skaarj1989, 2023-08-21:
Try the [GSM](https://github.com/skaarj1989/mWebSockets/tree/GSM) branch.

trickyau, 2023-08-21:
Slight improvement with that branch but still running into following. WiFI compiles fine, GSM does not.

```
c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.cpp: In member function 'void net::WebSocket::_send(uint8_t, bool, bool, const char*, uint16_t)':
c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.cpp:271:36: error: invalid conversion from 'char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
     bytesWritten += m_client.write(maskingKey, 4);
                                    ^~~~~~~~~~
In file included from c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM7600.h:25,
                 from c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:55,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\platform.h:72,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\utility.h:3,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.h:5,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.cpp:1:
c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmTCP.tpp:110:33: note:   initializing argument 1 of 'size_t TinyGsmTCP<modemType, muxCount>::GsmClient::write(const uint8_t*, size_t) [with modemType = TinyGsmSim7600; unsigned char muxCount = 10; size_t = unsigned int; uint8_t = unsigned char]'
     size_t write(const uint8_t* buf, size_t size) override {
                  ~~~~~~~~~~~~~~~^~~
c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.cpp:282:24: error: invalid conversion from 'const char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
         m_client.write(reinterpret_cast<const char *>(data), length);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM7600.h:25,
                 from c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:55,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\platform.h:72,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\utility.h:3,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.h:5,
                 from c:\Users\tricky\Documents\Arduino\libraries\mWebSockets\src\WebSocket.cpp:1:
c:\Users\tricky\Documents\Arduino\libraries\TinyGSM\src/TinyGsmTCP.tpp:110:33: note:   initializing argument 1 of 'size_t TinyGsmTCP<modemType, muxCount>::GsmClient::write(const uint8_t*, size_t) [with modemType = TinyGsmSim7600; unsigned char muxCount = 10; size_t = unsigned int; uint8_t = unsigned char]'
     size_t write(const uint8_t* buf, size_t size) override {
                  ~~~~~~~~~~~~~~~^~~
```

skaarj1989, 2023-08-21:
Can you give more information about your environment? IDE, compiler, c++ version, etc.?

trickyau, 2023-08-21:
I'm using plan old Arduino IDE version 2.1.1. Not sure if I makes any difference but TinyGSM is 0.11.7.

skaarj1989, 2023-08-21:
In my environment these errors are just warnings.
Just apply `reinterpret_cast<const uint8_t *>(...);`

skaarj1989, 2023-08-21:
AFAIK, After the handshake, a server cannot "assume" a client connection status (the only thing it can do is send a ping frame).
Are you sure that the server accepted the client handshake? What kind of server do you have?

skaarj1989, 2023-08-21:
Could you please check [this one](https://github.com/skaarj1989/mWebSockets/blob/eeeb71f06048b4a0807113df5fc302d044be5368/node.js/simple-server.js)?

closed 2023-09-29 by stale[bot]