skaarj1989/mWebSockets #68: Hardfault when computer disconnects and doesn’t properly terminate socket

gigaj0ule, 2023-12-17:
**Describe the bug**
Hardfault when computer disconnects and doesn’t properly terminate socket

**Environment info**
- IDE w/version STM32Duino
- Platform/Board: STM32
- Network controller (shield or module): W5500

**Expected behavior**
I have been experiencing an issue where, under the condition that the computer disconnects without properly closing the socket, if you attempt to call send(), we end up in the hardfault handler. I attempted to fix this issue by checking to see if &ws pointer was a valid address before calling send(), and yet we still have the issue. 

I found someone else who also had this issue too:

https://forum.arduino.cc/t/arduino-freezing-when-using-websocketserver/680272

Though they did not find a solution either

skaarj1989, 2023-12-21:
I'll try to reproduce your issue.
- Do you have a server or a client or your board?
- Does it occur on an included simple example?

closed 2024-01-28 by stale[bot]

gigaj0ule, 2024-01-28:
I think I figured out what was causing the problem.

In my program flow, when I detected a socket was closed, I closed the websocket object itself.

anticipating then, I would create another web socket, because the connection was dead.

As it turns out, you cannot do this directly in the closing call back because then you destroy the object the callback is a child of.

Instead, you must set a flag that the web socket is closed and then elsewhere in your program flow restart everything.

Causing nondeterministic behavior, because whenever a computer would close their connection, sometimes they would do it properly, or other times they would just orphan the connection

And because the closed call wouldn’t get triggered on an orphan connection, it would continue to work

So when the computer connection was I seeing this error

I don’t think it’s a fault of the library per se, but maybe we should include in the documentation somewhere that you shouldn’t modify the parent object in the closed connection callback.