skaarj1989/mWebSockets #69: Can not pass the verify with Arduino uno R4 wifi board in Arduino IDE.

samyaosheng, 2023-12-18:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment info**
- IDE w/version [Arduino IDE 2.2.1]
- Platform/Board: [Arduino Uno R4 Wifi]
- Network controller :  board embedded 

**Expected behavior**
Pass the verification at least.

**Failure log**
```
In file included from c:\Users\sam\Documents\Arduino\libraries\mWebSockets\src\base64\Base64.cpp:2:0:
c:\users\sam\documents\arduino\libraries\mwebsockets\src\platform.h:33:4: error: #error "Unsupported platform"
 #  error "Unsupported platform"
    ^~~~~
c:\Users\sam\Documents\Arduino\libraries\mWebSockets\src\base64\Base64.cpp:9:12: fatal error: pgmspace.h: No such file or directory
 #  include <pgmspace.h>
            ^~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

**Additional context**

skaarj1989, 2023-12-18:
Arduino UNO R4 support is available on the `master` branch.

samyaosheng, 2023-12-19:
> Arduino UNO R4 support is available on the `master` branch.

Thanks very much for your information!
R4 wifi looks pretty cool, although I've just finished testing the ESP32S3.

closed 2024-01-19 by skaarj1989