skaarj1989/mWebSockets #70: Client not working on ESP32-S3 WIFI

brucedk, 2024-01-21:
**Describe the bug**
I'm using a ESP32-S3 DevkitC-1 in wifi mode, when I'm working as a server, it all works, but when I set it as client, it gives me this error 
```
[  1265][E][esp32-hal-adc.c:158] __analogRead(): Pin 0 is not ADC pin!
```

**Environment info**
- mWebSockets version/branch: master
- IDE w/version: PlatformIO v3.3.2
- Platform/Board: ESP32-S3 DevkitC-1
- Network controller (shield or module): Wifi

**Failure log**
```
Connecting to tecnica
.. connected
Device IP: 192.168.10.236
[  1265][E][esp32-hal-adc.c:158] __analogRead(): Pin 0 is not ADC pin!
Connection failed!
```

brucedk, 2024-01-21:
I'm using the example codes to test

skaarj1989, 2024-01-21:
What kind of WebSocket server do you use?
Please mind that the SSL connection is not supported.

brucedk, 2024-01-21:
I'm using the same library with a arduino mega 2560 and a 28J60. with the arduino as a client and the esp32 as a server, it works. when I reverse, it give me the error above. the arduino as a server I can connect with a nodejs as a client. and I'm not using ssl

skaarj1989, 2024-01-21:
I'm talking about a scenerio when you have a WebSocket client (this library) uploaded into a ESP32 board.
What is your server endpoint? also this library?

open `config.h` and uncomment debugging defines.

skaarj1989, 2024-01-21:
And what about a debug log of a server?

skaarj1989, 2024-01-21:
The connection was rejected way before the handshake.
Make sure that your **ip** and **port** is correct when using `client.open(...)`

brucedk, 2024-01-23:
just check again, ip and port are correct. this error only happens when the ip and port are correct, if it is wrong it only shows the connection failed, without the analogread error

skaarj1989, 2024-01-23:
Actually it's the other way around, the client connets to a server (server can only accept/reject a client).
Anyways,  I can't help you with debugging without a servers log. Looks like WiFi controller rejects a connection (I don't know why).

On a board that you have a WebSocket client, could you upload any WebServer client sketch (try a built-in example)?

closed 2024-03-02 by stale[bot]