skaarj1989/mWebSockets #72: Unsupported frame size

infrafast, 2024-05-06:
Hello, using the code adapated for issue #58 ,

I got the below error message when trying to send the below Jsonframe from the browser to the esp32

Unsupported frame size = 476

followed by a:
[189085][E][WiFiClient.cpp:517] flush(): fail on fd 52, errno: 11, "No more processes"

presumably this is just because I need to increase a buffer somewhere, maybe you could help ?
Thanks
BR

{"command":"createModule","param":{"Module 2":{"possibleLayout":3,"currentLayout":0,"icon":"b","strips":[{"id":"ch2bus","index":0,"color":99,"icon":"Z","properties":["/ch/01/name","/bus/01/name"]},{"id":"ch2bus","index":1,"color":99,"icon":"Z","properties":["/ch/01/name","/bus/01/name"]},{"id":"ch2bus","index":2,"color":99,"icon":"Z","properties":["/ch/01/name","/bus/01/name"]},{"id":"ch2bus","index":3,"color":99,"icon":"Z","properties":["/ch/01/name","/bus/01/name"]}]}}}

skaarj1989, 2024-05-06:
Increase the buffer size (this information is in the [README](https://github.com/skaarj1989/mWebSockets?tab=readme-ov-file#configh))
https://github.com/skaarj1989/mWebSockets/blob/a4458add6b2e157581c298901a9f9a9a26f3090e/src/config.h#L30

closed 2024-05-09 by skaarj1989

skaarj1989, 2024-05-27:
There's no need for that, thank you though.