skaarj1989/mWebSockets #73: randomSeed calls always start at 0 on esp32

henkelis, 2024-05-26:
**Describe the bug**
On a Feather esp32s3, calling analogRead(0) results in an error and returns zero. That call is used for random seeds in the code.

**Environment info**
- mWebSockets version/branch: 1.6.0/master
- IDE w/version: Arduino IDE 2.3.2
- Platform/Board: Adafruit ESP32-S3 Feather with 4MB Flash 2MB PSRAM
- Network controller (shield or module): PoE Featherwing (W5500)

**Expected behavior**
A valid random value is required.

**Failure log**
```
[esp32-hal-adc.c:158] __analogReadRaw(): Pin 0 is not ADC pin!
```

**Additional context**

skaarj1989, 2024-05-26:
Thank you for reporting the issue.

You can work around this problem by heading into:
https://github.com/skaarj1989/mWebSockets/blob/a4458add6b2e157581c298901a9f9a9a26f3090e/src/WebSocket.cpp#L47
and replacing `0` with a correct `ADC` pin.

I don't have a `Feather esp32s3` board to test this. Could you check something like `A0`...`A5`?

henkelis, 2024-05-26:
Thanks. I can confirm that using `A0` works. I changed the same setting in WebSocketClient.cpp too.

closed 2024-05-26 by skaarj1989

paulkilroy, 2025-02-23:
A0 isn't defined on non-arduinos. Build breaks on ESP32...

infrafast, 2026-04-07:
hello from my perspective randomSeed(micros()); would be much better workaround than trying to use another analog read as pin setup is completely independant from socket sending

commit 99d1512e (2024-05-26): Fix #73