skaarj1989/mWebSockets #75: Broken examples/doc

dmtzs, 2024-06-25:
**Describe the bug**
Wrong examples out of date

you need to update the examples, the actual ones are useless

skaarj1989, 2024-06-25:
Everything compiles and works as it used to. Could you explain what you mean?

dmtzs, 2024-07-07:
I receive this errors when I copy your client code in teh README:
```
D:\github projects\Arduino\websocket\websocket.ino:13:17: error: 'net' is not a namespace-name
 using namespace net;
                 ^~~
D:\github projects\Arduino\websocket\websocket.ino:13:20: error: expected namespace-name before ';' token
 using namespace net;
                    ^
D:\github projects\Arduino\websocket\websocket.ino:20:23: error: variable or field 'onEventsCallback' declared void
 void onEventsCallback(WebsocketsEvent event, String data) {
                       ^~~~~~~~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:20:23: error: 'WebsocketsEvent' was not declared in this scope
D:\github projects\Arduino\websocket\websocket.ino:20:23: note: suggested alternative: 'WebSocketClient'
 void onEventsCallback(WebsocketsEvent event, String data) {
                       ^~~~~~~~~~~~~~~
                       WebSocketClient
D:\github projects\Arduino\websocket\websocket.ino:20:53: error: expected primary-expression before 'data'
 void onEventsCallback(WebsocketsEvent event, String data) {
                                                     ^~~~
D:\github projects\Arduino\websocket\websocket.ino:77:20: error: variable or field 'handleMessage' declared void
 void handleMessage(WebsocketsMessage message) {
                    ^~~~~~~~~~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:77:20: error: 'WebsocketsMessage' was not declared in this scope
D:\github projects\Arduino\websocket\websocket.ino:15:24: error: variable or field 'onMessageCallback' declared void
 void onMessageCallback(WebsocketsMessage message) {
                        ^~~~~~~~~~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:15:24: error: 'WebsocketsMessage' was not declared in this scope
D:\github projects\Arduino\websocket\websocket.ino:20:23: error: variable or field 'onEventsCallback' declared void
 void onEventsCallback(WebsocketsEvent event, String data) {
                       ^~~~~~~~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:20:23: error: 'WebsocketsEvent' was not declared in this scope
D:\github projects\Arduino\websocket\websocket.ino:20:23: note: suggested alternative: 'WebSocketClient'
 void onEventsCallback(WebsocketsEvent event, String data) {
                       ^~~~~~~~~~~~~~~
                       WebSocketClient
D:\github projects\Arduino\websocket\websocket.ino:20:53: error: expected primary-expression before 'data'
 void onEventsCallback(WebsocketsEvent event, String data) {
                                                     ^~~~
D:\github projects\Arduino\websocket\websocket.ino: In function 'void setup()':
D:\github projects\Arduino\websocket\websocket.ino:57:3: error: 'client' was not declared in this scope
   client.onOpen([](WebSocket &ws) {
   ^~~~~~
D:\github projects\Arduino\websocket\websocket.ino:57:3: note: suggested alternative: 'llrint'
   client.onOpen([](WebSocket &ws) {
   ^~~~~~
   llrint
D:\github projects\Arduino\websocket\websocket.ino:57:20: error: 'WebSocket' has not been declared
   client.onOpen([](WebSocket &ws) {
                    ^~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino: In lambda function:
D:\github projects\Arduino\websocket\websocket.ino:59:8: error: request for member 'send' in 'ws', which is of non-class type 'int'
     ws.send(WebSocket::DataType::TEXT, message, strlen(message));
        ^~~~
D:\github projects\Arduino\websocket\websocket.ino:59:13: error: 'WebSocket' has not been declared
     ws.send(WebSocket::DataType::TEXT, message, strlen(message));
             ^~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino: In function 'void setup()':
D:\github projects\Arduino\websocket\websocket.ino:61:21: error: 'WebSocket' has not been declared
   client.onClose([](WebSocket &ws, const WebSocket::CloseCode code,
                     ^~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:61:42: error: 'WebSocket' does not name a type
   client.onClose([](WebSocket &ws, const WebSocket::CloseCode code,
                                          ^~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino: In lambda function:
D:\github projects\Arduino\websocket\websocket.ino:64:5: error: expected '{' before ';' token
   });
     ^
D:\github projects\Arduino\websocket\websocket.ino: In function 'void setup()':
D:\github projects\Arduino\websocket\websocket.ino:64:5: error: expected ')' before ';' token
D:\github projects\Arduino\websocket\websocket.ino:65:23: error: 'WebSocket' has not been declared
[... 4 lines cut ...]
                                            ^~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino: In lambda function:
D:\github projects\Arduino\websocket\websocket.ino:68:5: error: expected '{' before ';' token
   });
     ^
D:\github projects\Arduino\websocket\websocket.ino: In function 'void setup()':
D:\github projects\Arduino\websocket\websocket.ino:68:5: error: expected ')' before ';' token
D:\github projects\Arduino\websocket\websocket.ino: In function 'void loop()':
D:\github projects\Arduino\websocket\websocket.ino:74:3: error: 'client' was not declared in this scope
   client.listen();
   ^~~~~~
D:\github projects\Arduino\websocket\websocket.ino:74:3: note: suggested alternative: 'llrint'
   client.listen();
   ^~~~~~
   llrint
D:\github projects\Arduino\websocket\websocket.ino: At global scope:
D:\github projects\Arduino\websocket\websocket.ino:77:20: error: variable or field 'handleMessage' declared void
 void handleMessage(WebsocketsMessage message) {
                    ^~~~~~~~~~~~~~~~~
D:\github projects\Arduino\websocket\websocket.ino:77:20: error: 'WebsocketsMessage' was not declared in this scope

exit status 1

Compilation error: 'net' is not a namespace-name
```

dmtzs, 2024-07-07:
Using Arduino uno R4 WIFI @skaarj1989

skaarj1989, 2024-07-07:
Now it's pretty obvious. Either you did not install **mWebSockets** or have more than one library with a header **WebSocketClient.h**, and **Arduino** picks the wrong one.

dmtzs, 2024-07-07:
Well I installed this library first when I installed the arduino IDE, also is a new PC and of course I installed the mwebsockets library

dmtzs, 2024-07-10:
Just a quick update, apparently I'm on the correct way, still having some issues but let you know when I'm finally able to subscribe me to a topic.
For now I'm having issues with that but will see

dmtzs, 2024-07-13:
Now it compiles and doesnt appear any error but is not connected to the websocket

dmtzs, 2024-07-13:
This is my code:
```ino
#include "WiFiS3.h"
#include <ArduinoJson.h>
#include <ArduinoHttpClient.h>
#include "arduino_secrets.h"
#include <WebSocketClient.h>
#include "functions.h"

char ssid[] = SECRET_SSID;  // your network SSID (name)
char pass[] = SECRET_PASS;
int status = WL_IDLE_STATUS;
// const uint16_t websockets_server_port = 8000; // Cambiar por tu puerto
// String websocket_path = "/ws/"; // Cambiar por tu ruta, asegúrate de incluir el token
using namespace net;
WebSocketClient client;

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    while (true);
  }
  
  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }
  
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, pass);
    delay(10000);
  }
  
  printWifiStatus();
  printWifiStatusGraphics();

  client.onOpen([](WebSocket &ws) {
    const char message[]{ "Hello from Arduino client!" };
    ws.send(WebSocket::DataType::TEXT, message, strlen(message));
  });
  client.onClose([](WebSocket &ws, const WebSocket::CloseCode code,
                   const char *reason, uint16_t length) {
    // ...
  });
  client.onMessage([](WebSocket &ws, const WebSocket::DataType dataType,
                     const char *message, uint16_t length) {
    // ...
  });

  client.open("ws://0.0.0.0/ws/<token>", 80);
}

void loop() {
  client.listen();
}

```
But is not working, I create also a python client to connect me to the same websocket and it worked, just this arduino code is not working

skaarj1989, 2024-07-13:
> if (!client.open("ws://localhost", 8000, "/ws/<token>")) 

Try using an IP address (localhost is unknown to Arduino) and do not use a prefix like [ws](https://github.com/skaarj1989/mWebSockets/blob/7c1efc973e05e9644c5490671fc480679b32c503/src/WebSocketClient.h#L23)

> Also planning to use wss and not ws

SSL is not supported.

dmtzs, 2024-07-13:
Understood so should be something liek this?:
`if (!client.open("<ip device>", 8000, "/ws/<token>"` I already tried that too 😢

skaarj1989, 2024-07-13:
IP and port of a server you'd like to connect to.

dmtzs, 2024-07-14:
I'm using the IP in which the websocket server is running and of course using the port in which the same websocket is running too.
About uncomment the debugging macros will show me more description of the error connection?

skaarj1989, 2024-07-14:
> I'm using the IP in which the websocket server is running

What is your server implementation (node.js, python, something else?).
If you try to connect to HTTPS server, the connection will fail.

> About uncomment the debugging macros will show me more description of the error connection?

Yes, in the serial monitor.

dmtzs, 2024-07-15:
Ok, so The only error I receive is this one: `Error in connection establishment: net::ERR_CONNECTION_REFUSED`
I'm on the same network and also I tried to connect me with the websocket using postman from a different computer and it worked (image attached) 😢 
[image]

Also about your first question my implementation is through python, code is next:
```python
from fastapi import FastAPI, WebSocket, WebSocketDisconnect
from fastapi.security import HTTPAuthorizationCredentials
from auth import create_token, get_current_user
from models import Message
import json

app = FastAPI()

class ConnectionManager:
    def __init__(self):
        self.active_connections: list[WebSocket] = []
        self.topics = {"topic1": [], "topic2": []}

    async def connect(self, websocket: WebSocket):
        await websocket.accept()
        self.active_connections.append(websocket)

    def disconnect(self, websocket: WebSocket):
        self.active_connections.remove(websocket)
        for topic in self.topics.values():
            if websocket in topic:
                topic.remove(websocket)

    async def send_personal_message(self, message: str, websocket: WebSocket):
        await websocket.send_text(message)

    async def broadcast(self, message: str, topic: str):
        for connection in self.topics[topic]:
            await connection.send_text(message)

    def subscribe(self, websocket: WebSocket, topic: str):
        if websocket not in self.topics[topic]:
            self.topics[topic].append(websocket)

manager = ConnectionManager()

@app.post("/token")
async def login():
    token = create_token({"username": "user1"})
    return {"access_token": token}

@app.websocket("/ws/{token}")
async def websocket_endpoint(websocket: WebSocket, token: str):
    user = get_current_user(credentials=HTTPAuthorizationCredentials(scheme="Bearer", credentials=token))
    await manager.connect(websocket)
    try:
        while True:
            data = await websocket.receive_text()
            message = json.loads(data)
            msg_obj = Message(**message)
            manager.subscribe(websocket, msg_obj.topic)
            await manager.broadcast(f"{user['username']}: {msg_obj.content}", msg_obj.topic)
    except WebSocketDisconnect:
        manager.disconnect(websocket)

if __name__ == "__main__":
    import uvicorn
    uvicorn.run(app, host="0.0.0.0", port=8000)

```

Also the code is tested and working if I create another client to connect me to the websocket and also from postman from the websocket section that postman has I was able to connect me successfully 😢

skaarj1989, 2024-07-16:
The error is triggered from [here](https://github.com/skaarj1989/mWebSockets/blob/7c1efc973e05e9644c5490671fc480679b32c503/src/WebSocketClient.cpp#L39)
As you can see, this is before the **WebSocket** logic comes in.

Try using a minimal echo server (as simple as possible).

dmtzs, 2024-07-16:
Well it doesn't matter, will try to find another solution, if I need to delete the authentication the is useless for me and also I tried with that but is not working.
No worries, thank you for the help

closed 2024-07-16 by dmtzs