skaarj1989/mWebSockets #76: WebSocket client is not connecting to any server ( online ) not local

buffed-ets, 2024-09-03:
WebSocket client is not connecting to the actual server ( echo.websocket.org )
but i can connect to local node server 

I'm using Ethernet module W5500 with [Mega 2560 PRO MINI]

here is my code logs after uploading it,
its attempting to connect to echo.websocket.org but fails!

[image]

here is another attempt trying to connect it to a local server ( node ) 

[image]

as you can see its connecting and sending data properly 

here is my code 

-------------------------------------------------------------
#include <WebSocketClient.h>
#include <ArduinoJson.h>
using namespace net;

byte mac[]{0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

WebSocketClient client;

int Light1 = 1;
int Light2 = 0;
int gas1 = 22.3;
void setup() {
  Serial.begin(115200);
  Ethernet.init(53);
  Ethernet.begin(mac);
  Serial.print(F("Device IP: "));
  Serial.println(Ethernet.localIP());
  client.onOpen([](WebSocket &ws) {
  Serial.println(F("Connected"));

    JsonDocument Home_status;
    Home_status["rooms"]["room1"]["Light"] = Light1;
    Home_status["rooms"]["room1"]["ac"] = Light2;
    Home_status["rooms"]["room1"]["gas"] = gas1;
    Home_status["rooms"]["room2"]["Light"] = Light1;
    Home_status["rooms"]["room2"]["ac"] = Light2;
    Home_status["rooms"]["room2"]["gas"] = gas1;
    char json_string[256];
    serializeJson(Home_status, json_string);

     ws.send(WebSocket::DataType::TEXT, json_string, strlen(json_string));
     Serial.print("msg sent");
    }
    );
   client.onMessage([](WebSocket &ws, const WebSocket::DataType dataType,
                      const char *message, uint16_t length) {
  switch (dataType) {
    case WebSocket::DataType::TEXT:
      Serial.print(F("Received: "));
      Serial.println(message);
      break;
    case WebSocket::DataType::BINARY:
      Serial.println(F("Received binary data"));
      break;
    }
    ws.send(dataType, message, length); // Echo back to server
  });
  client.onClose([](WebSocket &, const WebSocket::CloseCode, const char *,
                   uint16_t) {Serial.println(F("Disconnected\n")); });
  
  if (!client.open("echo.websocket.org", 80,"/")) {
  
    Serial.println(F("Connection failed!"));
    while (true);
    }
}

void loop() {
client.listen();
    }

-------------------------------------------------------------

my goal is to get it to connect to an online server kindly help me.

skaarj1989, 2024-09-07:
If I remember correctly `echo.websocket.org` was closed a few years ago.

buffed-ets, 2024-09-12:
i have tried my own server:
tserver-production.up.railway.app
it works when i used postman 

[image]

it didn't work Aswell 
"mind that WebSocket Secure is not supported"
is there a way to be able to connect to secure websocket ?

closed 2024-10-29 by stale[bot]